Ultimaker/Cura #4962: HELP link for OnLine Documentation does NOT work

Artistica1990, 2018-12-11:
https://ultimaker.com:8080/en/manuals/software

ChrisTerBeke, 2018-12-11:
Why is port 8080 used in this URL, it should juts be ultimaker.com/en/manuals/software.

Artistica1990, 2018-12-11:
The port 8080 part is added when the web site does not respond.
 
This is the original link:
http://ultimaker.com/en/support/software
 
Then we get below screenshot.
 

 
Marco Margaritelli
Artistica – Deruta of Italy
 <http://www.artistica.com/> www.artistica.com |  <http://www.derutaofitaly.com/> www.derutaofitaly.com |  <http://www.derutacandles.com> www.derutacandles.com 

Artistica Italian Gallery, Inc.
1874 Johns Drive – Glenview, IL 60025  | Tel.: (847) 730-3783 - Fax: (847) 730-5978

A family owned company since 1990
 <https://www.facebook.com/Artistica.Italian.Ceramics/>   <https://www.instagram.com/artistica_italian_ceramics>    <https://www.pinterest.com/artisticaig/>    <https://twitter.com/ArtisticaIG>

nallath, 2019-02-06:
Link responds just fine for me. Closing this.

closed 2019-02-06 by nallath