Ultimaker/Cura #4968: [3.6.0] Cura does not block automatic system sleep/shutdown while USB printing

TheThief, 2018-12-11:
**Application Version**
3.6.0

**Platform**
Windows

**Printer**
Any

**Steps to Reproduce**
Set Windows or other OS to automatically sleep after a few minutes, then begin a USB print.
Alternatively, Windows update performs an automatic restart mid-print.

**Actual Results**
PC sleeps mid-print, ruining print.

**Expected results**
PC should not automatically sleep during a print.

**Additional Information**
On Windows, the WinAPI functions `ShutdownBlockReasonCreate()/...Destroy()` and/or `SetThreadExecutionState(ES_CONTINUOUS | ES_SYSTEM_REQUIRED | ES_AWAYMODE_REQUIRED)` should help.

mahtDFR, 2020-03-27:
Thanks for the suggestion, we've decided to defer it as our products don't use serial printing these days, so we won't be putting development time into it. If anyone would like to contribute a fix, we're open to pull requests.

fieldOfView, 2020-03-27:
There is a plugin on the Marketplace (https://marketplace.ultimaker.com/app/cura/plugins/thopiekar/CuraPowerManagementPlugin). I don't know if it works on Windows.

closed 2026-09-06 by GregValiant