Ultimaker/Cura #4973: [3.5 - 3.6] Gradual support in Air at flat surfaces

TimoErdmann, 2018-12-11:
**Application Version**
3.6

**Platform**
Windows 10 - 64bit

**Printer**
custom FFM printer
[CFFFP_TestSupport.3mf.zip](https://github.com/Ultimaker/Cura/files/2668096/CFFFP_TestSupport.3mf.zip)

**Steps to Reproduce**
Option1: import FFM 
Option2: import an STL with flat bottom surface which needs support
**Actual Results**
[image]
The generated **gradual** support hangs in air. It will wall down and built up. But there is a possibillity that the extruder grabs these strings and fuses them with the part.

**Expected results**
[image]
expand the gradual support with a defined angle so it builds up to the layer underneath.

**Additional Information**
[image]

JohnEdwa, 2018-12-12:
Another possible implementation that would work fairly well with almost no additional material cost would be using bridging. Cura even has the necessary functionality already - simply generate a new 'support first layer grid' structure between the gradual infill steps using the new support density.

[image]

TimoErdmann, 2018-12-13:
Yes you could use bridging. But not all Materials are cabable of this feature and this then this will fail tremendously. So there can be an option with bridging but there has to be on option for the angled version too.