Ultimaker/Cura #4974: [Suggestion] OctoPrint API: Add Timer before print commences.

Thisismydigitalself, 2018-12-12:
Thank you for Cura 3.6 - So far best version I use daily.

for the last couple of months or so i started using PTG plugin for octoprint which is far more accurate in predicting Print time. Cura ETA doen not take into considuration Printer's EEPROM settings and PTG calculate and factor these in so i am very happy with the results which are few minutes off for few hours print whereas Cura ETA is sometimes off by an hour - %% wise this is useless.
PTG plugin needs few seconds to do its thingy and calculate and it stops the moment Print starts, hence my following uggestions. I hope this can be added to Cura as I did not get an answer from PTG side.

 
**Application Version**
3.6.0

**Platform**
Win10/64/NV/SSD

**Printer**
CFFFP

**Steps to Reproduce**
upload gcode to printer using Cura / OctoPrint API in 

**Actual Results**
When "Automatically start print" is checked i need for my printer to wait a given seconds before print commences. i don't know if Cura's OP plugin can do this, or this is on Octoprint side to take of this.

**Expected results**
would like to have the option to chose how many seconds before uploading till print starts.

fieldOfView, 2018-12-12:
Requests for the OctoPrint Connection plugin should go to this issue queue:

https://github.com/fieldofview/OctoPrintPlugin/issues

Thisismydigitalself, 2018-12-12:
Done. This issue can now be closed.

fieldOfView, 2018-12-12:
You can close it, I cannot.