Ultimaker/Cura #4984: .3mf.3mf in the filename

zhou13, 2018-12-13:
**Application Version** 3.6.0

**Platform** Linux

**Printer** Not related

**Steps to Reproduce**
Client menu File -> Save

**Actual Results**
Filename should be `Untitled.3mf`.

**Expected results**
Filename should be `Untitled.3mf.3mf`.

nallath, 2018-12-18:
Unfortunately, the file dialog sometimes does this on it's own. It's outside of our control what actually happens in this case.

closed 2018-12-18 by nallath

zhou13, 2018-12-18:
I don't quite understand why this is out of the control. If the file dialog automatically add `.3mf`, then you can just pass `Untitled` rather than the `Untitled.3mf` to the function call.  I think this is a bug that should be fixed so please at least leave the issue open.