Ultimaker/Cura #4987: Suggestion: Flip model

Pixopax, 2018-12-13:
Hi, it would be great to have a flip model function, as I often have to flip one model upside down. I can do that with rotate, I know, but a simple "flip", horizontal and vertical, would be nice to have. Should be eay to do as it is a 180° rotation.

fieldOfView, 2018-12-13:
The Mirror Tool does that.

AbeFM, 2018-12-13:
I submit that the "mirror" tool mirrors, making it upside-down, but also backwards.

fieldOfView, 2018-12-13:
The Mirror tool has 3 axes. If you think it is backwards, also flip it on the front-to-back axis.

I either don't understand your original request, or your reply.

AbeFM, 2018-12-13:
The "cool kids" have some features like "set face on bed" or other gui-based orientation tools. One slicer I've used shows you flats that you can click on to make that side down, based on some pre-analyis when you start the tool.

[image]

fieldOfView, 2018-12-13:
In that case, "flipping" can be done by using the mirror tool twice.

closed 2026-09-06 by GregValiant