Ultimaker/Cura #4990: Cura 3.6.0 Doesn't save or load material profiles.

Ruen1991, 2018-12-13:
**Application Version**
3.6.0

**Platform**
Windows 10 64bit

**Printer**
Creality Ender 3

**Steps to Reproduce**
Attempted to create custom material profile.

**Actual Results**
Cura does not save or load material profile. but can export and import, but profile does not appear when imported.

**Expected results**
Cura does not save or load material profile. but can export and import, but profile does not appear when imported.

**Additional Information**

Ruen1991, 2018-12-13:
Reinstalled Cura, works now.

theWaldschrat, 2019-02-23:
I'd like to reopen this issue, because I just ran into the same problem. Reinstalling Cura didn't help, either. Maybe deleting all settings might, but I'd like not to do that, because I've changed quite a lot.
Anything I can do to help solving that? Provide log-files or something like that?

When changing filament diameter the following ends up in stderr.log:
```
2019-02-23 17:13:57,902 - DEBUG - [MainThread] cura.Machines.Models.NozzleModel._update [34]: Updating NozzleModel.
2019-02-23 17:13:58,069 - INFO - [MainThread] cura.Machines.MaterialManager._updateMaps [279]: Updating material lookup data ...
2019-02-23 17:13:58,147 - INFO - [MainThread] cura.Settings.MachineManager._updateQualityWithMaterial [1226]: Updating quality/quality_changes due to material change
2019-02-23 17:13:58,147 - DEBUG - [MainThread] cura.Settings.MachineManager._updateQualityWithMaterial [1233]: Current quality type = [draft]
2019-02-23 17:13:58,147 - INFO - [MainThread] cura.Settings.MachineManager._updateQualityWithMaterial [1247]: Current available quality type [draft] is available, applying changes.
2019-02-23 17:13:58,158 - DEBUG - [MainThread] cura.Machines.Models.CustomQualityProfilesDropDownMenuModel._update [15]: Updating CustomQualityProfilesDropDownMenuModel.
2019-02-23 17:13:58,161 - DEBUG - [MainThread] cura.Machines.Models.QualityProfilesDropDownMenuModel._update [50]: Updating QualityProfilesDropDownMenuModel.
2019-02-23 17:13:58,197 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [352]: file:///C:/Program Files/Ultimaker Cura 3.6/resources/qml/Preferences/Materials/MaterialsPage.qml:236:13: QML MaterialsList: Binding loop detected for property "height"
2019-02-23 17:13:58,199 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [352]: file:///C:/Program Files/Ultimaker Cura 3.6/resources/qml/Preferences/Materials/MaterialsList.qml:135:5: QML Column: Binding loop detected for property "height"
2019-02-23 17:13:58,280 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [352]: file:///C:/Program Files/Ultimaker Cura 3.6/resources/qml/Preferences/Materials/MaterialsSlot.qml:44: TypeError: Cannot read property 'root_material_id' of null
2019-02-23 17:13:58,282 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [352]: file:///C:/Program Files/Ultimaker Cura 3.6/resources/qml/Preferences/Materials/MaterialsSlot.qml:44: TypeError: Cannot read property 'root_material_id' of null
2019-02-23 17:13:58,284 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [352]: file:///C:/Program Files/Ultimaker Cura 3.6/resources/qml/Preferences/Materials/MaterialsSlot.qml:44: TypeError: Cannot read property 'root_material_id' of null
2019-02-23 17:13:58,286 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [352]: file:///C:/Program Files/Ultimaker Cura 3.6/resources/qml/Preferences/Materials/MaterialsSlot.qml:44: TypeError: Cannot read property 'root_material_id' of null
2019-02-23 17:13:58,288 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [352]: file:///C:/Program Files/Ultimaker Cura 3.6/resources/qml/Preferences/Materials/MaterialsSlot.qml:44: TypeError: Cannot read property 'root_material_id' of null
2019-02-23 17:13:58,290 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [352]: file:///C:/Program Files/Ultimaker Cura 3.6/resources/qml/Preferences/Materials/MaterialsSlot.qml:44: TypeError: Cannot read property 'root_material_id' of null
2019-02-23 17:13:58,292 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [352]: file:///C:/Program Files/Ultimaker Cura 3.6/resources/qml/Preferences/Materials/MaterialsSlot.qml:44: TypeError: Cannot read property 'root_material_id' of null
2019-02-23 17:13:58,294 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [352]: file:///C:/Program Files/Ultimaker Cura 3.6/resources/qml/Preferences/Materials/MaterialsSlot.qml:44: TypeError: Cannot read property 'root_material_id' of null
2019-02-23 17:13:58,296 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [352]: file:///C:/Program Files/Ultimaker Cura 3.6/resources/qml/Preferences/Materials/MaterialsSlot.qml:44: TypeError: Cannot read property 'root_material_id' of null
2019-02-23 17:13:58,298 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [352]: file:///C:/Program Files/Ultimaker Cura 3.6/resources/qml/Preferences/Materials/MaterialsSlot.qml:44: TypeError: Cannot read property 'root_material_id' of null
2019-02-23 17:13:58,300 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [352]: file:///C:/Program Files/Ultimaker Cura 3.6/resources/qml/Preferences/Materials/MaterialsSlot.qml:44: TypeError: Cannot read property 'root_material_id' of null
2019-02-23 17:13:58,302 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [352]: file:///C:/Program Files/Ultimaker Cura 3.6/resources/qml/Preferences/Materials/MaterialsSlot.qml:44: TypeError: Cannot read property 'root_material_id' of null
2019-02-23 17:13:58,304 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [352]: file:///C:/Program Files/Ultimaker Cura 3.6/resources/qml/Preferences/Materials/MaterialsSlot.qml:44: TypeError: Cannot read property 'root_material_id' of null
2019-02-23 17:13:58,306 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [352]: file:///C:/Program Files/Ultimaker Cura 3.6/resources/qml/Preferences/Materials/MaterialsSlot.qml:44: TypeError: Cannot read property 'root_material_id' of null
[... cut ...]
2019-02-23 17:13:58,368 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [352]: file:///C:/Program Files/Ultimaker Cura 3.6/resources/qml/Preferences/Materials/MaterialsPage.qml:236:13: QML MaterialsList: Binding loop detected for property "height"
2019-02-23 17:13:58,370 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [352]: file:///C:/Program Files/Ultimaker Cura 3.6/resources/qml/Preferences/Materials/MaterialsList.qml:135:5: QML Column: Binding loop detected for property "height"
```

Ghostkeeper, 2019-02-27:
I think the greatest help then would be to take the configuration folder (that stderr.log is contained in), zip that all up and post it here. If that is too difficult or data-sensitive, then you could also try saving a project file (File -> Save...) and post that instead. Maybe that allows us to reproduce the problem.

theWaldschrat, 2019-02-27:
I hope that there is nothing sensitive in the configuration, at least there is nothing I could think of. So, here we go...or not, it's too large...and files named .001 aren't supported, either...so, renamed. Change them back to cura.zip.001, ...002, ...003, ...004, then unpack.
I'm using 3.6.0 but a while ago I installed (and then removed) 4.0 for testing. I don't know if that's part of the problem, so I also added the 4.0 config as found in the cura directory.

[cura.001.zip](https://github.com/Ultimaker/Cura/files/2911523/cura.001.zip)
[cura.002.zip](https://github.com/Ultimaker/Cura/files/2911524/cura.002.zip)
[cura.003.zip](https://github.com/Ultimaker/Cura/files/2911525/cura.003.zip)
[cura.004.zip](https://github.com/Ultimaker/Cura/files/2911526/cura.004.zip)

closed 2020-06-20 by no-response[bot]

Ghostkeeper, 2020-06-30:
We have since discovered that this issue depends on the locale of the user. In some locales, you need to use a comma instead of a period.