Ultimaker/Cura #4991: [Feature Req] Speed: Small Perimeters

CCS86, 2018-12-14:
Due to the motion control limitations of essentially all 3D printers, handling acceleration and jerk becomes increasingly difficult on smaller radii. Especially on small holes (short perimeters / walls), speeds which work very well on larger loops, are problematic. STL tolerancing means that the segment to segment angles get exaggerated, requiring more jerk and higher acceleration. This results in poor printed form.

The simple solution is to just do them at lower speed. Developing a feature (like Slic3r has) to evaluate the length of loops, and below a certain threshold, enact a slower (defined) printing speed, would be super useful!

smartavionics, 2018-12-14:
Hello @CCS86 , that capability already exists in my Cura releases which you can find at https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0

Thisismydigitalself, 2018-12-14:
Always wondered about this issue. 
I will definatly give this version a try.

CCS86, 2018-12-14:
@smartavionics, very cool!

Is there a reluctance to merge that feature into the main release branch?

smartavionics, 2018-12-14:
They (the Cura guardians) had the opportunity to merge this feature for several months but they didn't do it. I get bored keeping PRs up to date with the changes to the master code base so if PRs don't get any attention in a few months I close them. You can lead a horse to water, ...

Thisismydigitalself, 2018-12-15:
@smartavionics 
In regards to small features, Your version works like a charm.

Much Obliged.

smartavionics, 2018-12-15:
You're welcome.

DDDirk, 2018-12-15:
A problem with changing the print speed is that the extrusion speed (feed rate) also changes. But with a bowden tube, there is a delay between the feeder and what comes out the nozzle. So if the print speed slows down, there will initially be too much pressure in the nozzle, which will cause overextrusion. Especially with a small feature, that might be a problem. Except if it is printed after a travel move. The pressure may have oozed away, and you can actually get underpressure and underextrusion. Of course the two may also cancel out.

CCS86, 2018-12-15:
That's what linear advance and coasting are for. Avoiding all print speed changes, isn't a solution for bowden induced lag. 

Cura already has like 9+ print speed settings. One more for small loops would be a real benefit.

Thisismydigitalself, 2018-12-15:
I have only direct drive printers. 
Couple of hours ago i used @smartavionics version to print a model with small round holes & small text and for the first time ever I saw my printhead print those slowly and gracefully that i am now hooked to this feature. instead of printing these holes like blazingly fast they came out perfect.

AbeFM, 2018-12-18:
I have used a similar feature on other slicers and I miss it every time I come back.

CCS86, 2019-01-15:
Devs?

KampfGurke20, 2019-01-24:
@smartavionics
What must i do to get RepetierIntegration working with your Version?
I Downloaded the Plugin into "C:\Program Files\Ultimaker Cura master\plugins\RepetierIntegration" as i did with the Original 3.6, but there is No Button to Setup the Printer :(
[image]
[image]

fieldOfView, 2019-01-24:
Don't download plugins into the program files folder, but put them in the Configuration folder instead. See Help -> Show configuration folder, and then find the plugins folder contained in that.

While you are in the Configuration folder, locate the file named "Cura.log". It should show the reason why the plugin is not loading.

KampfGurke20, 2019-01-24:
I had tried both, first in the program folder and then in the configuration folder!
(Thats the Plugin: https://github.com/VMaxx/RepetierIntegration ) 
Because it has a Master Branch i thought it should work :(

So I can only wait until the plugin gets an update, right?
I think I'm too stupid to repair that myself.

Well, as long as I have to upload the GCode by hand.

Thanks for your help! :)

CCS86, 2019-01-24:
Guys, this seems unrelated to the feature request.

smartavionics, 2019-01-24:
It's true, we digress...

diegopradogesto, 2019-01-28:
Discussed internally with our materials and processing team and it seems that it's a nice feature to have. However we think it's also good to add **acceleration** and **jerk** as separate settings for the small features.

For internal reference: CURA-6158

Thisismydigitalself, 2019-03-11:
@smartavionics 
Hi Mark, Any chance your small featuers solution can be added to 3.6 release version?
I don't see myself moving to 4.x any time soon and would like to stick with 3.6 as i had some issues with Master versions.

Cheers.

fieldOfView, 2019-03-11:
New features will never be added to old/current versions of Cura. In other words: there will not be a version of Cura 3.6 with new features, unless they can be added in a plugin.

Thisismydigitalself, 2019-03-11:
I really hope this so invaluable feature will one day become a plugin and the sooner - the better.

smartavionics, 2019-03-11:
Hi @Thisismydigitalself , I'm afraid I can't make a 3.6+ release now. Sorry about that.

Thisismydigitalself, 2019-03-11:
:(

JorgeUliana, 2019-04-06:
Hi @smartavionics.
Thank you very much for creating this feature.
I was searching for him some time.
They should include it urgently in version 4.0, because to me this seems like a failure in Cura.
I did not have time to print the piece using its version, but in the simulated mode I noticed a series of improvements in the path between the printed parts and the printed parts.
I used the enable ironing, and the percussion on it also got better.

smartavionics, 2019-04-07:
Hello @JorgeUliana , this feature was presented to the Cura devs about 10 months ago and they ignored it for so long that I gave up keeping it up to date WRT the master branch. I have dusted it off and made it available again. As I already have around a dozen contributions (PRs) that are also being ignored, I wouldn't expect this to be available in Ultimaker Cura anytime soon. Let's be optimistic, they may surprise us yet.

SonOfSardaar, 2019-07-13:
I want this badly please, cura is not handling small holes good enough

AlexPl81, 2020-07-05:
> Hello @JorgeUliana , this feature was presented to the Cura devs about 10 months ago and they ignored it for so long that I gave up keeping it up to date WRT the master branch. I have dusted it off [...]

Thanks for the release you have uploaded. I can't believe such vital feature is missing. I have it on Prusa Slicer, and my Prusa MK3  prints small holes well, however my Ender 3, for which i use Cura makes quite a mess racing through 2mm holes at full speed. I started looking for solution (i didn't want to switch to Prusa Slicer for everything) and here it is. Thanks again.

Ghostkeeper, 2020-07-07:
This is added to Cura a few releases ago.

closed 2020-07-07 by Ghostkeeper