Ultimaker/Cura #5000: "Collecting Data" popup since v3.6.0

lowlyocean, 2018-12-15:
This annoying popup has been appearing every time I launch Cura,  and I have to click the X each time. Cura doesn't seem to remember my selection between runs.
[image]

lowlyocean, 2018-12-18:
So will Cura collect statistics even when I click the X? Why isn't Deny an option?

nallath, 2018-12-18:
You can disable it in the preferences menu. Since it's anonymized data, we don't even need to mention (or ask) this in the first place, but we decided to be transparent about it. 

It constantly asking is indeed a bug. What version are you running it on (I do recall that we fixed some things there some time ago, but i'm not a 100% sure)?

lowlyocean, 2018-12-19:
I'm using 3.6.0 and see no place to disable in preferences menu. Can you provide screenshot?

lowlyocean, 2018-12-19:
[image]
Found it, but it's already as I would expect it should be to disable the popups...

Thisismydigitalself, 2018-12-19:
in terms of bandwidth the data you send out is close to nothing.
in return you get a free slicer and still, you don't have to share anything.

thisiskeithb, 2018-12-19:
This bug is still present in Cura 4.0.

This notification is present at launch with the only options being Allow or close:
[image]

This setting is disabled:
[image]

My firewall doesn't show any traffic to stats.ultimaker.com in recent months, but it's confusing and annoying to dismiss that popup when I've disabled the feature.

nallath, 2018-12-19:
I'll look into it!

nallath, 2018-12-19:
That's weird. With 4.0 I also have a "more info" link sitting next to the allow button. The message keeps popping up untill you pressed the allow button (once you do, it's gone for good).

Liger0, 2018-12-23:
I can confirm I have always that popup.

lowlyocean, 2019-03-24:
This issue is still present in 4.0.0 @LipuFei

nallath, 2019-04-01:
This is fixed for 4.1

closed 2019-04-01 by nallath