Ultimaker/Cura #5001: [master] crash when loading existing project file

smartavionics, 2018-12-15:
**Application Version**

master

**Platform**

Linux

**Printer**

Kossel Mini (mb)

**Steps to Reproduce**

Select project .3mf file for loading.

**Actual Results**

Cura crashes while loading project.

**Expected results**

Project loads and we make it to the sunlit uplands where the unicorns live.

**Additional Information**

I fixed the problem by doing this:

````
diff --git a/plugins/3MFReader/ThreeMFWorkspaceReader.py b/plugins/3MFReader/ThreeMFWorkspaceReader.py
index 9ee2ef0dd..55296979b 100755
--- a/plugins/3MFReader/ThreeMFWorkspaceReader.py
+++ b/plugins/3MFReader/ThreeMFWorkspaceReader.py
@@ -794,7 +794,8 @@ class ThreeMFWorkspaceReader(WorkspaceReader):
             # Clear all existing containers
             quality_changes_info.global_info.container.clear()
             for container_info in quality_changes_info.extruder_info_dict.values():
-                container_info.container.clear()
+                if container_info.container:
+                    container_info.container.clear()
 
             # Loop over everything and override the existing containers
             global_info = quality_changes_info.global_info
````

LipuFei, 2018-12-18:
Hi @smartavionics , thanks for the fix. I've applied it to 4.0 dff9a3dfb

smartavionics, 2018-12-18:
You're welcome.

closed 2018-12-18 by smartavionics