Ultimaker/Cura #5010: Got Cura to build on Gentoo, but it crashes on startup

salfter, 2018-12-18:
I'm trying to get Cura to build on Gentoo so that I don't have to keep using the appimage, but I first ran into issues with v3.2.1.  I'm trying again now with v3.6.0...found a few new dependencies and took care of those, but it's still segfaulting on startup.

**Application Version**
3.6.0

**Platform**
Gentoo Linux

**Qt**
5.11.1

**PyQt**
5.10.1

**Display Driver**
x11-drivers/xf86-video-amdgpu-18.1.0

**Steps to Reproduce**
```
sudo layman -a salfter # this is my overlay, with Cura and dependencies
sudo emerge -av cura
/usr/bin/cura # load my build, not the appimage in /usr/local/bin
```

**Actual Results**
```
[MainThread] UM.Resources.__initializeStoragePaths [414]: Initializing storage paths
[MainThread] UM.Resources.__initializeStoragePaths [425]: Config storage path is /home/salfter/.config/cura/master
[MainThread] UM.Resources.__initializeStoragePaths [433]: Data storage path is /home/salfter/.local/share/cura/master
[MainThread] UM.Resources.__initializeStoragePaths [445]: Cache storage path is /home/salfter/.cache/cura/master
[MainThread] UM.PackageManager.__init__ [49]: Found bundled packages JSON file: /usr/share/uranium/resources/bundled_packages/uranium.json
[MainThread] UM.PackageManager.__init__ [49]: Found bundled packages JSON file: /usr/share/cura/resources/bundled_packages/cura.json
[MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [103]: Trying OpenGL context 4.1...
[MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [119]: Trying OpenGL context 2.0...
[MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [128]: We got at least OpenGL context 2.0: 3.1 No profile
[MainThread] UM.Qt.QtApplication.initialize [138]: Detected most suitable OpenGL context version: 2.0 No profile
[MainThread] UM.Qt.QtApplication.initialize [144]: Initializing job queue ...
[MainThread] UM.Qt.QtApplication.initialize [148]: Initializing version upgrade manager ...
[MainThread] UM.Settings.ContainerRegistry.addContainer [371]: Container [empty_definition_changes] added.
[MainThread] UM.Settings.ContainerRegistry.addContainer [371]: Container [empty_variant] added.
[MainThread] UM.Settings.ContainerRegistry.addContainer [371]: Container [empty_material] added.
[MainThread] UM.Settings.ContainerRegistry.addContainer [371]: Container [empty_quality] added.
[MainThread] UM.Settings.ContainerRegistry.addContainer [371]: Container [empty_quality_changes] added.
[MainThread] UM.PackageManager._loadManagementData [83]: Loaded bundled packages data from /usr/share/uranium/resources/bundled_packages/uranium.json
[MainThread] UM.PackageManager._loadManagementData [83]: Loaded bundled packages data from /usr/share/cura/resources/bundled_packages/cura.json
[MainThread] UM.PackageManager._loadManagementData [105]: Loaded user packages management file from /home/salfter/.local/share/cura/master/packages.json
[MainThread] UM.PackageManager._saveManagementData [117]: Package management file /home/salfter/.local/share/cura/master/packages.json was saved
[MainThread] UM.PluginRegistry.initializeBeforePluginsAreLoaded [92]: Loading plugin configuration file '/home/salfter/.config/cura/master/plugins.json'
2018-12-17 10:41:50,726 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin ConsoleLogger
2018-12-17 10:41:50,742 - INFO - [MainThread] CuraEngineBackend.CuraEngineBackend.__init__ [80]: Found CuraEngine at: /usr/bin/CuraEngine
2018-12-17 10:41:50,742 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin CuraEngineBackend
2018-12-17 10:41:50,747 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin UpdateChecker
2018-12-17 10:41:50,747 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [378]: Plugin ConsoleLogger was already loaded
2018-12-17 10:41:50,750 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin LocalContainerProvider
2018-12-17 10:41:50,750 - INFO - [Thread-1] UpdateChecker.UpdateCheckerJob.run [36]: Checking for new version of cura
2018-12-17 10:41:50,753 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin SimpleView
2018-12-17 10:41:50,755 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin LocalFileOutputDevice
2018-12-17 10:41:50,756 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin FileLogger
2018-12-17 10:41:50,759 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin ScaleTool
2018-12-17 10:41:50,760 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin SelectionTool
2018-12-17 10:41:50,762 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin MirrorTool
2018-12-17 10:41:50,763 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin CameraTool
2018-12-17 10:41:50,766 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin RotateTool
2018-12-17 10:41:50,768 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin TranslateTool
2018-12-17 10:41:50,771 - WARNING - [MainThread] STLReader.STLReader.<module> [26]: Could not find numpy-stl, falling back to slower code.
2018-12-17 10:41:50,772 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin STLReader
2018-12-17 10:41:50,774 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin STLWriter
2018-12-17 10:41:50,775 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin OBJReader
2018-12-17 10:41:50,777 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin OBJWriter
2018-12-17 10:41:50,777 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [378]: Plugin UpdateChecker was already loaded
[... 186 lines cut ...]
    self._bootstrap_inner()
  File "/usr/lib64/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/usr/lib64/python3.6/threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/lib64/python3.6/site-packages/UM/Backend/Backend.py", line 163, in _storeStderrToLogThread
    self._backendLog(line)
  File "/usr/lib64/python3.6/site-packages/UM/Backend/Backend.py", line 94, in _backendLog
    Logger.log("d", "[Backend] " + line_str.strip())
  File "/usr/lib64/python3.6/site-packages/UM/Logger.py", line 68, in log
    logger.log(log_type, message)
  File "/usr/lib/uranium/plugins/FileLogger/FileLogger.py", line 44, in log
    self._logger.debug(message)
Message: '[Thread-10] UM.Backend.Backend._backendLog [94]: [Backend] [ERROR] Connection reset by peer'
Arguments: ()
Exception in thread Thread-10:
Traceback (most recent call last):
  File "/usr/lib64/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/usr/lib64/python3.6/threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/lib64/python3.6/site-packages/UM/Backend/Backend.py", line 160, in _storeStderrToLogThread
    line = handle.readline()
OSError: [Errno 9] Bad file descriptor
```

MathyV, 2018-12-19:
Hi, I was bitten by the same bug, this seems to be related to the version of PyQt5. Recently 5.9.2 was removed from the tree but Cura doesn't play nice yet with 5.10+

I was able to fix it by manually installing 5.9.2. You can find the ebuild here:
https://github.com/gentoo/gentoo/blob/f035df035f8640ed7ea4685105f0e7dd21aef51d/dev-python/PyQt5/PyQt5-5.9.2.ebuild

Also, there's a bug open in bugzilla talking about this:
https://bugs.gentoo.org/673220

PS: Cura is actually in the portage tree, and quite actively maintained, so your effort to create ebuilds would be more than welcome over there. Currently I have a pull request running for 3.6.0:

https://github.com/gentoo/gentoo/pull/10595

salfter, 2018-12-19:
Copied that older PyQt5 into my overlay...it's building now.  There aren't a huge number of packages that depend on it, and none depend on v5.10, so I'm hoping that'll be enough to get it running.

Last time I looked for Cura in Portage, it was way behind.  I had my own working Cura ebuilds up to around v3.2.1, at which point it quit working and I switched over to using the appimage just to get it running again.  v3.6.0 has been out a while now, but the latest Cura ebuild in Portage is for v3.4.1.  I'm using v3.6.0 on Windows at the office, so it'd be nice to have the same version running everywhere.

FWIW, my ebuild (and its dependencies) are at https://gitlab.com/salfter/portage, and it's installable through Layman.

salfter, 2018-12-19:
Looks like it's up and running now...got to the "add printer" dialog, which never happened before.  Sweet!

closed 2018-12-19 by salfter

zwim, 2019-04-06:
Hello Salfter,
I have added your overlay, cura-3.6.0 is building well, but if I try to emerge cura-4.0.0 portage is complaining about a missing file:

>>> Verifying ebuild manifests

!!! A file listed in the Manifest could not be found: /var/lib/layman/salfter/media-gfx/curaengine/files/curaengine-4.0.0-build-fix.patch

Could you please add the patch to your overlay.

Thanks