Ultimaker/Cura #5019: Can't copy/paste text into setting

BagelOrb, 2018-12-19:
**Application Version**
Cura 4.0.0-BETA

**Platform**
Linux Mint 19 Cinnamon
(Cinnamon 3.8.9)
(Linux kernel 4.15.0-38-generic)

**Printer**
UM3

**Steps to Reproduce**
Copy text. e.g. "10.0592231765546"
Select any setting value in Cura
Ctrl+A
Ctrl+V

**Actual Results**
Setting text is selected, but pasting has no effect

**Expected results**
Setting value is pasted

**Additional Information**
This is reproducible 30% of the time, but when it happens it keeps on happening - a restart is required.
10% of the time it's already a problem from startup. Sometimes there's no problem the first 10 times you paste a setting value and then all of a sudden I cannot do it anymore and I need to restart Cura.

When doing structural tests it is important to keep track of your settings and structurally test different setting values. The range of values for a setting I want to test is saved in a separate file from which I want to copy/paste settings.

Now I'm restarting Cura 10 times with no luck of being able to paste on the first try.
This is starting to become quite annoying!

BagelOrb, 2018-12-20:
So you _can_ reproduce it. The text in the reproduction steps has more than 4 decimal places.

nallath, 2018-12-21:
Apart from that it works like that all the time for me (thus making it perhaps a bit more "This works as intended, but we might need to change the intention" vs it being a bug)

BagelOrb, 2018-12-21:
I swear that it worked for me some times, but maybe when I copy a value from LibraOffice Calc it doesn't always copy the full value or something.

TheKrush, 2022-01-10:
Could we get the number of decimal spaces increased to 4? MPSM magic numbers are 4 decimals

https://github.com/metaxis/mpsm-cura-profile

Ghostkeeper, 2022-01-18:
Cura outputs g-code coordinates in micrometres, so having the number of decimal spaces increased, at least for the layer height, would have no immediate effect.

closed 2026-09-06 by GregValiant