Ultimaker/Cura #5023: CURA 3.6.0 z hop over printed part not working

dagnabitboy, 2018-12-19:
Cura 3.6.0, Windows 10 pro, A stock CR-10s and a DIY CR-10S clone  (with Marlin 1.1.9) both have the problem.  On this particular print the z-hop over printed parts is not working.  As the print gets taller you can hear a loud click as the head runs into an edge of the print.  I have z hop enabled, which seems to work well on other prints.  It's just this one print it seems to be having trouble with.  If you watch the on screen emulation you can see it too.  As the print is getting built, there are 4 legs (screen shot attached).  The hotend moves from the leg on the right to the leg on the left.  The hotend travels right over the leg on the left, reaches the center of that leg, then raises the z axis (increments the layer number in the emulation).  You can see this behavior in all the layers, but any layer from 100 up to 200 is a good place to watch.  I would have expected the hotend to z hop over the left leg, then drop back down to it's starting point for that leg.
[pyramid.3mf.zip](https://github.com/Ultimaker/Cura/files/2696629/pyramid.3mf.zip)
[image]

smartavionics, 2018-12-19:
Hello @dagnabitboy , please save the cura project, zip and attach to this issue. The file you attached above only contains the model and not the settings which I need to check out your problem. Thanks.

dagnabitboy, 2018-12-19:
OK, sorry, I goofed!  Here's what you need. (I hope!)
[pyramid.zip](https://github.com/Ultimaker/Cura/files/2696906/pyramid.zip)

closed 2018-12-19 by dagnabitboy

smartavionics, 2018-12-20:
Thanks for the project file. If you turn off the Z Hop Only Over Printed Parts setting it will hop when moving between the corner areas. Also I recommend turning on the Optimize Wall Printing Order setting. Hope this helps!

dagnabitboy, 2018-12-20:
Thanks very much for looking into this for me!  I will try the change in settings you suggest and see what happens.  I'll let you know.