Ultimaker/Cura #5025: [master] simple model becomes segmented

smartavionics, 2018-12-20:
**Application Version**

master branch

**Platform**

Linux

**Printer**

Any

**Steps to Reproduce**

Load the model below and slice.

**Actual Results**

The layer view shows that one end of the model is segmented into a "string of sausages" but a similarly curved region on the other end of the model is not segmented.

**Expected results**

The wall outline is not segmented.

**Additional Information**

Most issues like this are normally caused by a bad model but, so far, I cannot find a problem with this model. S3D can slice it OK, freecad says that it has no issues.

[40x10.zip](https://github.com/Ultimaker/Cura/files/2697867/40x10.zip)

[image]

[image]

smartavionics, 2018-12-20:
I found a workaround, using just a tiny amount of horizontal expansion joins the segments.

[image]

Perhaps Cura should add a similarly small amount by default although that would require doing an offset on every mesh.

nallath, 2018-12-21:
Does the issue also occur with older versions of Cura (3.6 / 3.5?)

smartavionics, 2018-12-21:
3.4.1 doesn't have the problem so it's a regression introduced since that release.

Ellecross, 2019-01-03:
Hey guys, I see it happen still in 3.4.1. 
Then in 3.3 and 3.2, it looked even worse. 
I have a feeling it has always been this way? :/ 

See images below: 
[image]
[image]
[image]

smartavionics, 2019-01-03:
Hi @Ellecross , you're absolutely right, I was mistaken to say earlier that 3.4.1 did not have the problem.

(Maybe) interestingly, I have discovered that if you set the wall line width such that it only prints with thin walls enabled, the thin wall isn't segmented...
[image]

smartavionics, 2019-01-04:
I think this is caused by a non-zero Outer Wall Inset, setting that to zero gives OK results.

[image]

Closing...

closed 2019-01-04 by smartavionics

PaulD79, 2019-09-23:
I solved it by changing to a different print orientation.
PaulD79