Ultimaker/Cura #5028: Profile's 'STAR' symbol makes me feel uncomefortable

Thisismydigitalself, 2018-12-20:
**Application Version**
3.6, 4.0B

**Platform**
Win10/64

**Printer**
CFFFM

**Steps to Reproduce**
Open a profile and then edit any setting(s)

**Actual Results**
I love working with Cura. Cura has so many settings. that's a good thing, but..
Once you touch the settings, the profile is now flagged with a 'star' meaning your current setting(s) differs from saved profile. this is annoying. i never feel safe to start a print when i see this star cause go figure what changed from the original profile. to top that, Cura remembers these changes and when you restart Cura clean - you get this star.  i need a button to open this same compare window we have when we load a differnet profile which asks us if we want to keep new settings or run over them so it will be clear and safe to start a few hours print job with some peace in mind.

More than I need new UI,  I need ease of use.
I was expecting 4.0 to bring better workflow and functionality and in return the feeling is going backwards.  I know that my words might sound too hard but please believe me that this is my constructive criticism from someone that works daily with Cura and wants to see it getting better.

Thisismydigitalself, 2018-12-20:
Related:
https://github.com/Ultimaker/Cura/issues/4932

nallath, 2018-12-20:
Well, that's a first that we hear someone about that functionality (it has been there since 2.1 or 2.3 I believe) ;-)
The idea was that you can view the changes in preferences -> profiles. There it shows a list of changes compared to the stored profile.

GuiltyPixel, 2018-12-22:
I agree that the star is awkward. It feels very Web Last.0 and it's kind of ambiguous. I would rather know at a glance what has changed, if it is stored under a name, and be able to do List <--> List style editing in case I wanted to keep my settings but swap one thing for example, it would save time. Can haz new thing?

Thisismydigitalself, 2018-12-26:
> The idea was that you can view the changes in preferences -> profiles. There it shows a list of changes compared to the stored profile.

@nallath 
Each small change to my saved profile flags the profile with a star. this is a big issue. 
I asked before and I am asking again - there must be a real-time summery of our selected settings within the profile at all times. no way around this. out of so many settings i want to know before sending gcode to printer that my selected setting which i decided that are important for my printed part properties are not effected in a bad way. there is no way around this.