Ultimaker/Cura #5030: there is no option for dual extruders and only one nozzel

surfettekw, 2018-12-20:
**Application Version**
3.6

**Platform**
Windows

**Printer**
Geeetech A10M

**Steps to Reproduce**
Anytime I try to print from extruder 2 Cura doesn't heat up element because it is looking for element 2 but I only have one shared nozzle. 

**Actual Results**
Bed heats, element does not and it tries to print

**Expected results**
Would like a way to make it so that Cura can understand two extruders one nozzle.

**Additional Information**

nallath, 2018-12-21:
Well, no Ultimaker printer has this, so it's not very likely that any of the UM devs are going to spend time on fixing this. 
Even if we would, we also don't have a machine to test it with.

antman2, 2018-12-22:
@surfettekw Have you tried printing the model with extruder 2 and just a skirt with extruder 1?

surfettekw, 2018-12-22:
I will try that again when I am not so short on time :)

Sent from XFINITY Connect Mobile App

------ Original Message ------

From: antman2
To: Ultimaker/Cura
Cc:Kathi, Mention
Sent: December 22, 2018 at 6:51 AM
Subject: Re: [Ultimaker/Cura] there is no option for dual extruders and only
one nozzle (#5030)

@surfettekw(https://github.com/surfettekw)Have you tried printing the model with extruder 2 and just a skirt with extruder 1?

antman2, 2019-01-22:
@surfettekw I've had an opportunity to try this lately with a Tronxy X5S-2E which is a dual extruder single hotend setup like yours. The Tronxy runs Marlin 1.1.0-RC3 firmware which, unlike Marlin 2.x firmware, doesn't really understand multiple extruder single hotend printers.

When switching from the second to first extruder Cura sends `M109 T0 S{material_print_temperature}` followed by `M109 T1 S0`, but the second command turns off the shared heater so the hotend cools and eventually it is unable to extrude.

To keep the hotend hot when switching between filaments you need to edit the printer's profile to add either an M104 or M109 command to the second extruder's "Extruder Start G-code" but specify T0 (which is the first extruder as far as the printer is concerned).

Example 1:
```
M104 T0 S{material_print_temperature} ; Set 1st extruder temp, don't wait
```

Example 2:
```
M109 T0 S{material_print_temperature} ; Set 1st extruder temp, wait for target temp
```

This works for me when only wanting to print with the second extruder, too.

Ghostkeeper, 2020-07-27:
Newer versions of Cura have an option in the Printer Settings dialogue called "Shared Heater" to address this issue.

closed 2020-07-27 by Ghostkeeper

mirecta, 2021-02-07:
Shared Heater - where it is i have daily builds and it not found

fvrmr, 2021-02-09:
Hi @mirecta it is still there as machine setting, but not visible. You can find more here: #8148