Ultimaker/Cura #5032: Cura 3.6 marketplace crashes on Fedora 29

morozsm, 2018-12-20:
**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

nallath, 2018-12-21:
We've had reports about this before, it seems to have something to do with the SSL library / configurations that fedora changed in 29 (I'm still on 28 myself, where it does work)

morozsm, 2018-12-21:
Well, I might can help to debug the issue... Fedora 29 is becoming more and more popular :)

miriad, 2018-12-27:
Stuck on this myself. Can't use octoprint, cause I can't open the Marketplace. Happy to help debug (Am a dev myself, just not for cura)

nallath, 2019-01-03:
Good news! We managed to find a fix. It should work for the second beta of 4.0

closed 2019-01-03 by nallath

Stoccarda, 2019-01-04:
Have the same problem on Fedora 29.
Will it be fixed in 3.6.1 or do we have to wait for a stable 4.0?

Ellecross, 2019-01-04:
Hi @morozsm , @miriad and @Stoccarda 
I've a build here from today. It's also 4.0 Beta but it contains the fix. 
Can you give it a go and let me know if it still crashes for you? 
You can get the AppImage here: https://drive.google.com/open?id=1EG7qILT9c7LnHzSjenIlYDhrGk656xca

Let me know how it goes. ;)

Stoccarda, 2019-01-07:
Hi @Ellecross,
the click on Marketplace no more closes CURA :-)

Can this beta be already used for printing instead of 3.6.0, or better use 3.5.1?
I have missed the FreeCAD import filter in it (maybe other features missing too).
Generic ASA and PETG will also be great for coming versions.

Off this topic: in 3.6.0 the printing times after slicing were overlapped by "Save on removable media" button. This is also fixed by the different design.

The print open dialogue of 3.5.1. is easier to use on a PC with more than one drive

Ellecross, 2019-01-07:
Hi @Stoccarda, good to know it works for you! 

I would always recommend a newer version of Cura because of the fixes it contains. 

But if you have to work with a plugin that's not available yet for 4.0, then I'm afraid you'll have to wait - at least until the creator makes their plugin compatible for 4.0. :/ sorry

nallath, 2019-01-07:
The toolbox was renamed to marketplace.

Stoccarda, 2019-01-07:
Got it.
Was looking more for materials and plugins currently almost empty

Yannik, 2019-02-16:
Here is how to fix this for 3.6.0:
`sudo dnf install compat-openssl10`

Start cura like this:
`LD_PRELOAD="/usr/lib64/libssl.so.10 /usr/lib64/libcrypto.so.10" ./Cura-3.6.0.AppImage`