Ultimaker/Cura #5033: [3.6.0] Concentric Top/Bottom pattern does not print from outside to center of print

stefann82, 2018-12-20:
**Application Version** 
3.6.0

**Platform** 
Windows 7

**Printer** 
Mingda MD-6L
Saved files here: [test_cylinder.zip](https://github.com/Ultimaker/Cura/files/2700575/test_cylinder.zip)

**Steps to Reproduce**
1) Import cylinder model: https://www.thingiverse.com/thing:824730 (attached in above zip file)
2) Scale part by 5x (500%)
3) Set profile to Draft - 0.2mm
4) Wall thickness set to 1.2 mm, top/bottom pattern set to concentric, 10% infill.
5) Click prepare.

**Actual Results**
Select Layer view and set layer to the first top layer (layer 247 for me) laid on top of the infill and watch how the layer is printed. The layer prints the walls first, then the nozzle travels to the centre of the part and starts printing the top concentric (spiral) starting from the center and outwards towards the wall.

**Expected results**
Per the help website: https://ultimaker.com/en/resources/52343-shell, under section Top/bottom pattern:

_Concentric: The pattern is printed from the outside to the center of the print._

I expected the opposite of what was generated.

**Additional Information**
[image]

smartavionics, 2018-12-20:
Probably this that did it ...

https://github.com/Ultimaker/CuraEngine/commit/c86501427571f2d2879be7aa994e9b38615bd391

ryano88, 2019-01-10:
I have tried this in the following versions:
3.2.1, 3.3.1, 3.4.1, 3.5.1, 3.6 and 4.0 beta

Only in 3.2.1 does the concentric pattern correctly go from the outside of the print to the centre.  In all other versions listed, the print goes from centre to outside as the stefann82 said.

smartavionics, 2019-01-10:
That's not surprising given when the change I cited above was done...

[image]

That change was merged just after 3.2.1 was tagged.

othbert, 2019-02-05:
I thought I was going insane! Thanks for the report that it works properly in 3.2.1 will try that.

Kevin138, 2019-07-06:
Is there any update on this issue? Could it be a check box to change it from outside-in to inside-out instead of being forced to do it in inside-out direction?

I would think you would get better adhesion by going outside-in.

Ghostkeeper, 2019-07-08:
When going from outside to inside the blobbiness tends to ruin your adhesion. We've only seen positives from going from inside to outside so far.

If there is to be made a setting that configures the order in which concentric skin lines are printed, then there would need to be:
* A reason for the setting to be Inside-to-Outside in some cases and a reason for the setting to be Outside-to-Inside in other cases, depending on the model or the printer.
* Some non-zero chance that the average user is going to find this setting, understand what it does and change it.

I'm not seeing either of those, for this feature.

closed 2019-07-08 by Ghostkeeper

othbert, 2019-07-08:
The negative situation arose for me with a tall twisting tower structure that was hollow inside and had a platform at the top. 

Using supports was not viable as there was no clear path to the buildplate, so support would have had to be laid inside the hollow structure in order to provide a base for the platform. It would have contacted (or simply not been created) with the inner walls of the tower, in a location impossible to reach to remove it. The tower top platform printed with zig zag in the end, but as this was global, it affected the rest of the print.

In order to get this to work exactly as desired I had to cut the model into two separate prints (printed at the same time with per model settings) then attach them together.

Ghostkeeper, 2019-07-08:
In such a case I'd have used tree supports, probably :sunglasses:

othbert, 2019-07-08:
Tree supports would have reached from the buildplate to the platform, but would still have the consequence of being unable to access the platform base in order to remove the support properly and clean it up :)

mistic100, 2019-08-31:
Can this be reevaluated ? I have 100% print failure when trying concentric pattern because the material does not stick. It is not a problem when the patterns starts from a wall.

mistic100, 2019-08-31:
Also the documentation is still wrong ;-)

voxelbec, 2020-03-11:
For my use cases the inside-to-outside default for concentric patterns is almost always the wrong choice and it's very frustrating that this can not be overridden.

PLEASE make the concentric direction a user selectable option. It's the ONE bugbear that has prompted me to look at alternative slicers.

darkpoetry, 2020-08-13:
When filling in a space that would work fine using lines, concentric starts in the middle. Yes, one could put in supports, but why waste the plastic when the software could just as well detect that there's nothing beneath it for several layers and flip to outside-in behavior in that circumstance automatically?

Ghostkeeper, 2020-08-17:
That would make sense for when the skin is printed in mid-air. What's more, it would be even better if it were to try and print the lines more straightly in that case, because printing curves in mid-air is asking for trouble. Maybe it should switch to a lines pattern? Lines would reduce the distance that needs to be _bridged_.

darkpoetry, 2020-08-17:
Agreed. I haven't tried any experiments on bridging concentric from an outer edge inward but I imagine lines will work fine in all such cases.

Ghostkeeper, 2020-08-19:
So then your solution is to Enable Bridge Settings and perhaps adjust the threshold for when it starts bridging, with Bridge Skin Support Threshold?

darkpoetry, 2020-08-19:
I don't know if that would work. I think switching to lines is the most elegant solution to this issue.

Ghostkeeper, 2020-08-24:
Right, that's one of the things that the bridging routine does.

matthewatkins1, 2020-10-10:
Hi Everyone, 

I would also very much like to have the ability to print inside outward. 

I use a 2mm nozzle with a large volume Delta Printer and only print using a single outer wall. 

Having the ability to choose inside out concentric pattern direction would allow the entire print to circulate from the inside of the shape to the final layer providing an very smooth finish and removing the need for any retraction.

Ghostkeeper, 2020-10-12:
Cura will only print concentric patterns from inside outward, currently.

matthewatkins1, 2020-10-14:
> Cura will only print concentric patterns from inside outward, currently.

Thats true, but this is a feature request not a support request, ideally, this option would be added.

Ghostkeeper, 2020-10-16:
You requested the pattern to print from inside outward, and it does. There is no way to make it NOT do what you request. That's indeed what this ticket is about.

c24w, 2021-12-14:
This is definitely achievable by doing what @Coizado suggests.  You can force it to behave that way for all layers by enabling the _Outer Before Inner Walls_ option (which may have other consequences for your print).  There might be better ways to dupe Cura in this way, but here's an example of it working:

https://user-images.githubusercontent.com/710406/146014094-dc6495ef-a93d-4d2b-811d-1cab94a4e89e.mp4

Michal-Szczepaniak, 2022-01-22:
@c24w doesn't work in cura 4.9.0 at least for me :/ my issue is that i have holes in the model (going from top into model) and cura instead of doing the first layer from the outside in (so that it can attach to wall or infill) it goes from inside out making cura basically  print in thin air

matthewatkins1, 2023-07-01:
> This is definitely achievable by doing what @Coizado suggests. You can force it to behave that way for all layers by enabling the _Outer Before Inner Walls_ option (which may have other consequences for your print). There might be better ways to dupe Cura in this way, but here's an example of it working:
> 
>  concentric-outside-in.mp4

That doesn't work for Concentric Patterns when you have Connect Top and bottom Polygons selected

f412design, 2023-10-22:
I want to add that the behavior from 5.2.1 (outside to inside) is preferrable to me as well. When using smooth PEI or glass bed, going from out to in for concentric pattern will result in perfectly smooth surface for entire first layer. With the new behavior of inside to out (after walls), the connection from the inside to the walls is a visible seam.

I had never had the blobbing issue that this change was intended to solve. I wish there was a setting for this. Like others, I will probably stick with 5.2.1 until there is an option.

soxhi8, 2024-06-10:
Having an additional setting to control the concentric pattern order, without changing the overall wall order, would be quite useful.

EMatthews93, 2025-01-28:
> When going from outside to inside the blobbiness tends to ruin your adhesion. We've only seen positives from going from inside to outside so far.
>
> If there is to be made a setting that configures [...]

I am running into this issue right now, however as opposed to needing it for my bottom layer, I'm wanting it for my top layer.

My model is really large, and does not need a lot of structure, so to keep print time and costs down I am trying to minimize infill (down to 10%). For the top layers, however, I need the best way to print on low infill. I thought, "hey, I could probably use the wall to help support my top fill layer. This would be the perfect time for concentric fill!". However, the fill line goes exactly where I don't want it - starting right in the middle of the infill gaps.

Please add a setting.

Michal-Szczepaniak, 2025-01-28:
Thats's exact same thing I wanted to do. When concentric starts from inside it starts printing in the air and messes up the print while it could be outside in, starting from wall an get much better result.