Ultimaker/Cura #5034: [3.6.0] Setting for printing direction

rosdyana, 2018-12-21:
**Application Version**
3.6.0

**Platform**
Ubuntu 18.04

**Printer**
Custom Printer

**Issue/Problem**
I would like to change my printing direction become one same direction. Currently, it's run zig zag.

**Actual Results**
[image]

**Expected results**
I want it print on the same direction, ex: top to bottom and not continue in the middle.
Is there any configuration I can try to change the direction, or I should change something from the source in CuraEngine?

**Additional Information**
https://community.ultimaker.com/topic/25837-how-to-change-printing-direction/

rosdyana, 2018-12-21:
Hello @smartavionics , thank you for your response. Sorry if my question is not clear.
I made a slow version of my GIF file
[image]
as we can see on those GIF, it's run back and forth, is there any configuration that I can change to make it just print one direction like GIF bellow
[image]

about "not continue in the middle" ,
[image]
Is there any configuration to keep printing in green area instead of red area first, even it needs to cross the hole is ok for me.

Thanks a lot.

smartavionics, 2018-12-21:
Thanks for the further info. I think I now understand what you are asking. Unfortunately there is no means of telling Cura that you wish the skin lines to be drawn in one direction only from left to right. Sorry about that.

rosdyana, 2018-12-21:
@smartavionics , do you think it takes a lot effort if I try understand and modify the codes in CuraEngine ?

smartavionics, 2018-12-21:
Yes, it's a lot of effort. But tell me, why do you want to change how that skin is printed. What are you trying to achieve? Perhaps there is another way to achieve the same goal?