Ultimaker/Cura #5038: Cura 4.0.0 (and all other versions) keep closing when opening or saving a file.

Goop2526, 2018-12-21:
I can drag .stl files in and the software works smoothly. I can edit settings, slice the model, etc. However, whenever I do something involving the file explorer (file->open, exporting, or saving to file), cura closes. There is no error report, and I've tried about every fix I can find online. I've tried deleting the roaming data files, I've tried deleting shell extensions like Google Drive and Adobe, and I've tried reinstalling the software, rebooting my computer, etc. I can get Cura to work on my dell inspiron laptop, which runs windows 7, however I cannot get any version of Cura to work on my desktop.

**Application Version**
Cura 4.0.0 and all earlier versions 

**Platform**

Windows Edition: Windows 10 Home
Processor: Intel(R) Core(TM) i5-4210M CPU @ 2.60GHz 2.60 GHz
Installed Memory (RAM): 8.00GB
System Type: 64-bit Operating System, x64-based processor

**Printer**
 Monoprice Mini v1

**Steps to Reproduce**
Step 1: Launch Cura
Step 2: after cura starts, Click File->Open

Alternatively:
Step 1: Launch Cura
Step 2: Drag .stl file into Cura
Step 3: After slicing, click "Save to File" or "File->Save"

**Actual Results**
Cura opens the windows file explorer and then closes the entire application about 4-5 seconds later 

**Expected results**
Cura should open/save the file

**Additional Information**
Below is my log file
[stderr.log](https://github.com/Ultimaker/Cura/files/2702989/stderr.log)

alekseisasin, 2018-12-21:
Hmm, can you try to run Cura as Administrator ("Run as administrator")?

Goop2526, 2018-12-21:
Ran as administrator- still having the same issue.

om2kw, 2018-12-21:
works fine, can not replicate on OSX

Goop2526, 2018-12-22:
does not work fine on my windows 10 machine. Anyone able to help?