Ultimaker/Cura #5041: [3.6.0] Reducing infill below 18% reduces wall print speed

lekv, 2018-12-21:
**Application Version**
3.6.0

**Platform**
MacOS Mojave

**Printer**
This happens with the CR-10 and Ender 3 stock profiles without changes (except to the infill, see below).

**Steps to Reproduce**
Load [this 25mm cube](https://www.thingiverse.com/thing:2267549), set the infill density to something below 18%, e.g. 0%. Then look at the resulting STL.

**Actual Results**
The gcode prints outer walls with a speed lower than expected. The default (and my setting) is 30mm/sec (F1800), but the actual speed is lower (F1772.7 for 17% infill, F877.6 for 0% infill). Inner wall speed is affected proportionally.

**Expected results**
Walls should print with the configured speed. Wall speed should not depend on infill percentage.

**Additional Information**
Reducing the outer wall thickness to a single layer (0.4mm), the wall printing speed drops further to F600.

This is layer 95 with no infill:
```;LAYER:95
;MESH:cube.stl
G0 X161.9 Y161.9 Z19.2
M204 S500
M205 X20 Y20
;TYPE:WALL-INNER
G1 F1755.2 X138.1 Y161.9 E767.92223
G1 X138.1 Y138.1 E768.71382
G1 X161.9 Y138.1 E769.50541
G1 X161.9 Y161.9 E770.297
M204 S5000
M205 X30 Y30
G0 F7200 X162.3 Y162.3
M204 S500
M205 X20 Y20
;TYPE:WALL-OUTER
G1 F877.6 X137.7 Y162.3 E771.1152
G1 X137.7 Y137.7 E771.9334
G1 X162.3 Y137.7 E772.7516
G1 X162.3 Y162.3 E773.5698
M204 S5000
M205 X30 Y30
G0 F7200 X162.1 Y162.3
G0 X161.5 Y161.618
;MESH:NONMESH
G0 X161.9 Y161.9
;TIME_ELAPSED:1140.725710
```

This is with the default of 20% infill density:
```
;LAYER:95
;MESH:cube.stl
G0 X138.259 Y138.259 Z19.2
M204 S500
M205 X20 Y20
;TYPE:FILL
G1 F3600 X161.739 Y161.739 E1618.68523
G1 X161.809 Y161.809
M204 S5000
M205 X30 Y30
G0 F7200 X161.8 Y161.809
G0 X161.739 Y159.277
M204 S500
M205 X20 Y20
G1 F3600 X152.547 Y161.74 E1619.00174
M204 S5000
M205 X30 Y30
G0 F7200 X153.065 Y161.74
M204 S500
M205 X20 Y20
G1 F3600 X159.356 Y138.258 E1619.8103
G1 X159.381 Y138.162
M204 S5000
M205 X30 Y30
G0 F7200 X159.381 Y138.2
G0 X161.738 Y140.642
M204 S500
M205 X20 Y20
G1 F3600 X138.26 Y146.933 E1620.61873
M204 S5000
M205 X30 Y30
G0 F7200 X138.259 Y146.745
M204 S500
M205 X20 Y20
G1 F3600 X153.254 Y161.739 E1621.32402
G1 X153.324 Y161.809
M204 S5000
M205 X30 Y30
G0 F7200 X153.324 Y161.8
G0 X159.276 Y161.74
M204 S500
M205 X20 Y20
G1 F3600 X161.739 Y152.548 E1621.64054
M204 S5000
M205 X30 Y30
G0 F7200 X161.739 Y153.065
M204 S500
M205 X20 Y20
G1 F3600 X138.259 Y159.356 E1622.44903
G1 X138.163 Y159.381
M204 S5000
M205 X30 Y30
G0 F7200 X138.2 Y159.381
G0 X140.641 Y161.739
M204 S500
M205 X20 Y20
G1 F3600 X146.933 Y138.259 E1623.25753
M204 S5000
M205 X30 Y30
G0 F7200 X146.745 Y138.26
M204 S500
M205 X20 Y20
G1 F3600 X161.739 Y153.254 E1623.9628
G1 X161.809 Y153.324
M204 S5000
M205 X30 Y30
G0 F7200 X161.8 Y153.324
G0 X161.738 Y146.853
M204 S500
M205 X20 Y20
G1 F3600 X138.259 Y153.144 E1624.77126
G1 X138.163 Y153.169
M204 S5000
M205 X30 Y30
G0 F7200 X138.2 Y153.169
G0 X138.259 Y155.23
M204 S500
M205 X20 Y20
G1 F3600 X144.768 Y161.739 E1625.07743
G1 X144.838 Y161.809
M204 S5000
M205 X30 Y30
G0 F7200 X144.838 Y161.8
G0 X146.853 Y161.739
M204 S500
M205 X20 Y20
G1 F3600 X153.145 Y138.258 E1625.88596
G1 X153.17 Y138.162
M204 S5000
M205 X30 Y30
G0 F7200 X153.17 Y138.2
G0 X155.231 Y138.26
M204 S500
M205 X20 Y20
G1 F3600 X161.739 Y144.768 E1626.19208
G1 X161.809 Y144.838
M204 S5000
M205 X30 Y30
G0 F7200 X161.8 Y144.838
G0 X147.447 Y138.259
M204 S500
M205 X20 Y20
G1 F3600 X138.259 Y140.721 E1626.50845
G1 X138.163 Y140.746
M204 S5000
M205 X30 Y30
G0 F7200 X138.2 Y140.746
G0 X140.722 Y138.258
M204 S500
M205 X20 Y20
G1 F3600 X138.26 Y147.446 E1626.82483
G1 X138.235 Y147.542
M204 S5000
M205 X30 Y30
G0 F7200 X161.83 Y161.83
G0 X161.9 Y161.9
M204 S500
M205 X20 Y20
;TYPE:WALL-INNER
G1 F3600 X138.1 Y161.9 E1627.61642
G1 X138.1 Y138.1 E1628.40801
G1 X161.9 Y138.1 E1629.1996
G1 X161.9 Y161.9 E1629.99119
M204 S5000
M205 X30 Y30
G0 F7200 X162.3 Y162.3
M204 S500
M205 X20 Y20
;TYPE:WALL-OUTER
G1 F1800 X137.7 Y162.3 E1630.80939
G1 X137.7 Y137.7 E1631.62759
G1 X162.3 Y137.7 E1632.44579
G1 X162.3 Y162.3 E1633.26399
M204 S5000
M205 X30 Y30
G0 F7200 X162.1 Y162.3
G0 X161.5 Y161.618
;MESH:NONMESH
G0 X138.259 Y138.259
;TIME_ELAPSED:1320.078546
```

I'm attaching both gcode files and the Cura project.

[Archive.zip](https://github.com/Ultimaker/Cura/files/2703692/Archive.zip)

smartavionics, 2018-12-21:
That's because when using the lower infill density the layer doesn't take so long to print. Your profile is specifying a min layer time of 10 seconds. If you reduce the min layer time, the wall speed is not reduced.

[image]

lekv, 2018-12-21:
Thank you for the fast reply and for helping me. I dug around the CuraEngine source to learn how the speed gets computed and found the place where this gets applied. Apologies for not digging in the source longer.

closed 2018-12-21 by lekv