Ultimaker/Cura #5043: [3.6.0] a random layer has been assigned to the wrong z height.

megablue, 2018-12-22:
**Application Version**
3.6.0

**Platform**
windows 10 pro 64bit

**Printer**
my custom delta

**Steps to Reproduce**
slice https://www.thingiverse.com/thing:2493386 small moai at 0.01mm layer height
should be triggered by maximum resolution, i was running 0.02mm, it seems fine at 0.1mm

**Actual Results**
crash into a print and try to destroy my printer. 

**Expected results**
properly assign the layer height 

**Additional Information**
[image]

[CFFFP_moai-Small-NO_OVERHANG.3mf.zip](https://github.com/Ultimaker/Cura/files/2705523/CFFFP_moai-Small-NO_OVERHANG.3mf.zip)

smartavionics, 2018-12-22:
Just sliced your project and cannot see that weird line. Could you please zip up your gcode and attach to this issue. Thanks.

[image]

megablue, 2018-12-22:
I've lost the gcode. I just tried again, for some reasons, I couldn't reproduce it either... i swear i did re-sliced a few times and it was broken.

Ghostkeeper, 2019-05-28:
Indeed this looks like a duplicate of #3815. Since that issue was closed though, please continue discussion here: #5788.

closed 2019-05-28 by Ghostkeeper