Ultimaker/Cura #5046: [3.6.0] Cura Win10, infill pattern for bridging will not change

DominicBartel, 2018-12-23:
**Application Version**
3.6.0

**Platform**
Windows 10

**Printer**
Wanhao I3 Maker select (irrelevant)

**Steps to Reproduce**
Load file with bridge
Enable bridge settings
Enable "bridge has multiple layers"
Adjust infill to attempt to change pattern

**Actual Results**
infill will not change for bridge layers 2 and 3, all other infill changes as intended.

**Expected results**
bridge layers 2 and 3 infill to reflect infill settings

**Additional Information**

Overall I'm attempting to bridge using ABS and cranking the fan when bridging occurs. The first bridge layer usually comes out fine, however if it does not I cannot adjust bridge layers 2 and 3 to strand the filament from one side to the other. It only allows the standard zig zag pattern

Issue with 2nd/3rd layer:
[image]

First layer/intended for all layers:

[image]

.Gcode
[BridgeInfillIssue.zip](https://github.com/Ultimaker/Cura/files/2705691/BridgeInfillIssue.zip)

smartavionics, 2018-12-23:
Hello @DominicBartel , when I developed the bridging feature I did try orientating the 2nd bridge layer at 90 degs (and the third at 0 degs) WRT the direction of the bridge but I found that I consistently achieved better results in terms of bridge dimensions (width) and also layer adhesion. I believe printing the 2nd and 3rd layers at +/- 45 to the bridge direction is generally optimal.

CCS86, 2020-01-30:
I was about to open a new issue for the same reason, but found this one.

I agree with the OP that we should have control of the infill pattern for bridge layers 2 and 3. If the first layer doesn't join well, coming back at 45* will not help it recover.

Please add a setting to the subsequent bridge layers for infill pattern type.

smartavionics, 2020-01-30:
> If the first layer doesn't join well, coming back at 45* will not help it recover.

Agreed, but it begs the question:  why doesn't the first layer join well? It sounds to me as if the settings you are using for the first layer are not optimal. Please provide a project file that contains a bridge with a failed first layer.

CCS86, 2020-01-30:
Is it really difficult to imagine a scenario where successfully pulling every bridge strand across has less than 100% success rate?

Filament type, print temp, and span of the bridge present enough variability to cause a sub 100% success rate on first bridge layers.

Even just the heat and friction of a 45* layer on #2 could be enough to break the fragile strands on a long bridge.

I would think that starting angled infill at shallower angles like 15*, then increasing subsequently would be an interesting approach. 

Either way, giving the user control just seems best. It's impossible to speculate every possible scenario and how the print behaves.

smartavionics, 2020-01-30:
OK, let's try that one again, please provide a project file that includes a bridge whose first layer is not satisfactory. Thanks.

Ghostkeeper, 2020-02-03:
It's indeed impossible to speculate every possible scenario, but it's also impossible to add every conceivable way to adjust the g-code as a setting. Or rather, it is but then you're ending up with a text editor that directly writes the g-code. So besides giving the user control we also need to think about what we give them control over, and we'd like to give them control over things that will likely improve their print quality. The shallower angles seems like a better change to me, but it warrants some testing if it really works like that because that could also make the strands adhere less well horizontally.