Ultimaker/Cura #5047: [Request] Settings visibility export

Liger0, 2018-12-23:
Hi there.
There are much profile settings option so it requires much times to setup the visibility of the ones you need everytime, and then after resetting them for any reasons it a loss of time.
Would it be possible to implement an option to export the visibility settings in a file?
Thanks.

Ellecross, 2018-12-27:
Sounds like a nice idea. Thanks @Liger0, I'm going to raise this idea to the Cura team. If a ticket gets created as a new feature, I'll update you. ;)

Ellecross, 2019-01-03:
Ticket for a new feature created. 
Devs see CURA-6071

fvrmr, 2021-06-03:
We are constantly working on making Ultimaker Cura better. Our community is a big part of that by requesting new features and reporting bugs. 

To be able to focus on the most requested and needed features from our community and from Ultimaker, we have decided to remove this feature request from our backlog. We believe we will not work on this anytime soon and will therefore defer this issue.

Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.

closed 2023-07-13 by github-actions[bot]