Ultimaker/Cura #5055: change nozzle diam cura 4.0

cool0025, 2018-12-24:
**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

Ghostkeeper, 2018-12-24:
It's still in the printer settings (under the tab of the appropriate extruder). If you actually changed your hardware to have a different nozzle size, you should still be able to change it there.

If you didn't change any hardware, I advise you to just change the line width.

cool0025, 2018-12-24:
[image]
i dont find the place , can you  show me where they are now , before in this option , with the setting of the filament

Liger0, 2018-12-25:
Marketplace -> Printer settings

cool0025, 2018-12-26:
[image]
[image]
[image]
nothing in marketplace no printer setting here if you know where in cura beta 4.0 , take a picture and let me know , i have the setting for 1 printer with 0.6 nozzle , and 1 another model with 0.4 , ok 2 different model , but i swich between 3 king of nozzle , and i just saying i see nothing in this version of cura to change dhis setting , before present and easy to find .

fieldOfView, 2018-12-26:
The Printer Settings plugin has not yet been released for 4.0. I have submitted a compatible version but it hasn’t been sanctified yet.

cool0025, 2018-12-26:
haha , ty , i have reinstall 3.6 , and use both , i like the new 4.0 interface ,

closed 2018-12-26 by cool0025