Ultimaker/Cura #5056: Generating incomplete G Code.

Yari27, 2018-12-24:
**Application Version**
3.6.0

**Platform**
Windows 10 64 bit 1809

**Printer**
Anycubic i3 Mega 
[AI3M_2_caps.zip](https://github.com/Ultimaker/Cura/files/2706907/AI3M_2_caps.zip)

**Steps to Reproduce**
Layer Height = 0.2;
Infill = 20%;
Generate Support = true;
Build Plate Adhesion = true;
Support X/Y Distance = 1.2;
Export G code to card;
Print;

**Actual Results**
Only a stroke, a raft or just a few layers are generated. Some files can not even be opened in the program.

**Expected results**
It should generate g code for all (entire) models.

**Additional Information**
[cura.zip](https://github.com/Ultimaker/Cura/files/2706927/cura.zip)
[image]

smartavionics, 2018-12-24:
It sliced and the gcode looks OK - perhaps you have a problem saving to card? Try another card?

Yari27, 2018-12-26:
Yes, the problem is with the card.

GregValiant, 2023-09-14:
Can this be closed?

closed 2023-09-29 by github-actions[bot]