Ultimaker/Cura #5057: Nach Pause Druck ohne Filament

v98w, 2018-12-24:
Nach Pause at Height druckt der Drucker ohne Filamenteinzug weiter. Mit Cura 3.6 funktioniert alles bei gleichen Einstellungen. Drucker: Wanhao D6, letzte Firmware.

Ghostkeeper, 2018-12-24:
My limited high school German translation (please use English on the Internet):

> After Pause at Height, the printer continues without extruding. With Cura 3.6 and the same settings everything works. Printer: Wanhao D6, most recent firmware.

Perhaps you could show us the g-code that occurs around the pause? Maybe the Pause at Height plug-in has made a boo-boo.

v98w, 2018-12-27:
Hi Ghostkeeper,

thanks for your translation and your answer. After success with Cura 3.6 i deinstalled the beta version. I just want give a hint, perhaps i´m not the only one with this issue. If it´s interesting to the programmers i´ll try to reproduce this fault.

Best regards 

Wolfgang

closed 2020-06-20 by no-response[bot]