Ultimaker/Cura #5059: [3.6] Cura crashes on startup

Liger0, 2018-12-24:
**Application Version**
3.6

**Platform**
Windows 10 x64

**Printer**
Cr-10s dual extruder

**Steps to Reproduce**
Start Cura

**Actual Results**
It crashes

**Expected results**
It should start

**Additional Information**
[cura.zip](https://github.com/Ultimaker/Cura/files/2707525/cura.zip)

Ellecross, 2018-12-27:
Hey @Liger0 , yeah you're right. I see the crash error message you got in the logs: 

`2018-12-24 16:34:04,816 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [65]: An uncaught error has occurred!
2018-12-24 16:34:04,823 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]: Traceback (most recent call last):
2018-12-24 16:34:04,830 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:   File "X:\3.6\build\inst\lib\python3.5\site-packages\cura\Settings\ContainerManager.py", line 321, in clearUserContainers
2018-12-24 16:34:04,833 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:   File "X:\3.6\build\inst\lib\python3.5\site-packages\cura\Settings\MachineManager.py", line 859, in correctExtruderSettings
2018-12-24 16:34:04,837 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:   File "X:\3.6\build\inst\lib\python3.5\site-packages\cura\Settings\MachineManager.py", line 846, in getIncompatibleSettingsOnEnabledExtruders
2018-12-24 16:34:04,840 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]: TypeError: int() argument must be a string, a bytes-like object or a number, not 'SettingFunction'`

Can you tell me more about what you were doing when the crash happened? 
Alternatively, you can zip the \AppData\Roaming\cura folder so I can try and reproduce this. Maybe it is an issue, indeed. If it is, we have to see how to reproduce it so that we can fix it. 

With regards to slowness, we did some improvement and more tests on performance on the 4.0 beta version. The UI has changed as well, FYI. https://ultimaker.com/en/products/ultimaker-cura-software/download-request/209

Ellecross, 2018-12-27:
Oh wait, it looks like it's the same crash as your other ticket: https://github.com/Ultimaker/Cura/issues/5064 I checked the logs and it's the same CRITICAL error messages. 

Can you confirm?

closed 2018-12-27 by Liger0