Ultimaker/Cura #5062: Initial Layer Horizontal Expansion Not Doing Anything

Lukrative525, 2018-12-25:
**Application Version**
Cura 3.6.0

**Platform**
Windows 7 Professional 64-bit

**Printer**
My own printer (a rebuilt Anet A8), with machine settings based on Prusa i3 MK2

**Steps to Reproduce**
[x axis right z mount bug report.zip](https://github.com/Ultimaker/Cura/files/2709042/x.axis.right.z.mount.bug.report.zip)

I'm not sure if this is somehow a problem with my laptop, or if it's actually a bug with Cura, but all I had to do was to load any model, change the initial layer horizontal expansion to anything but what the normal horizontal expansion was, and then slice it.

**Actual Results**
When I did this, I could see in the preview (layer view) that the first layer was being offset by the same amount as the regular horizontal expansion, even though it was set to something different. The regular horizontal expansion had an obvious effect.

**Expected results**
You should be able to see a visible difference in size between the first layer and subsequent layers.

**Additional Information**
I have seen it work on my computer before with the same install, but within the last four or five days, this feature all of a sudden stopped working.

The attached zipped folder is a project file in which the expansions differ by 2mm. These are exaggerated values that I would never actually use.

smartavionics, 2018-12-25:
Hello @Lukrative525 , I played with your project and found that when the Slicing Tolerance is set to Middle, the bottom layer does have its own expansion but for the other values of Slicing Tolerance, the bottom layer appears to use the same expansion as the layers above (as you have found out).

I do not know whether this is a bug or if the creator of the slicing tolerance setting intended it to have this effect.

Leave this issue open and perhaps it will get picked up by one of the Cura devs and they may decide it's a bug and do something about it.

[image]

Lukrative525, 2018-12-27:
If anyone else is having this issue, you can get around it temporarily by saving the gcode twice, once with the horizontal expansion set to what you want it for the upper layers (which we will call the "upper layer gcode"), and once with the horizontal expansion set to what you want it for the bottom layer (which we will call the "first layer gcode"). Then just copy and paste the first layer from the "first layer gcode" in place of the first layer in the "upper layer gcode", and use a G92 at the start of the second layer to set the extruder position to what it was at the end of the first layer in the "upper layer gcode".

Note: the comments in the gcode call the layer touching the build platform layer 0, so the second layer would be layer 1.

Let's hope that the devs can figure out why the initial layer horizontal expansion doesn't work with anything but the "exclusive" setting for slicing tolerance.

Lukrative525, 2019-01-20:
I have several profiles with different layer heights. I use 0.15mm for the initial layers on all of them. It would seem that having the layer height set to the same as or less than the layer height for all subsequent layers also contributes to this issue. If the initial layer height is greater than the normal layer height, the initial layer expansion works, even using the "exclusive" slicing tolerance.

Ghostkeeper, 2020-04-21:
I'm unable to reproduce this issue in Cura 4.6. Was this fixed?

Lukrative525, 2020-04-22:
I don't even use Cura anymore, but after installing it, I also couldn't reproduce the problem. This issue appears to be fixed.

closed 2020-04-22 by Ghostkeeper