Ultimaker/Cura #5067: [3.6] Unneeded tree support

Liger0, 2018-12-26:
**Application Version**
3.6

**Platform**
Win10 x64

**Printer**
Cr-10s double extruder

**Steps to Reproduce**
Load the file

**Actual Results**
There are unneeded supports on the center

**Expected results**
There shouldn't be supports on the center

**Additional Information**
[image]
[unneeded supports.zip](https://github.com/Ultimaker/Cura/files/2711015/unneeded.supports.zip)

Ellecross, 2018-12-27:
I think this is expected because the model still has part of its mesh that needs support in the middle - thus you have the tree support in it. Support is also generated even if you choose the normal Cura support (and not the tree support). 

You can use the Support Blocker to prevent the support from being generated in that area. 

[image] I placed it on this part of the model. Scale 15x15x10.

Ghostkeeper, 2019-06-12:
This got fixed by @schneider007 through this pull request: https://github.com/Ultimaker/CuraEngine/pull/937

closed 2019-06-12 by Ghostkeeper