Ultimaker/Cura #5068: Toggling "Iron" does not cause need to re-slice

tlhintoq, 2018-12-27:
**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

tlhintoq, 2018-12-27:
@Ellecross 
Yep.  I see this _every_ time, _every_ model.
I have a screen-capture video with real-time clock showing it not changing after 2 minutes 30 seconds.  I can't attach an MP4, but if you want to see it just let me know how to send it to you.

smartavionics, 2018-12-27:
This happens a lot for me. I often have to hit F5 to reload the model(s), that makes the Prepare button active again.

tlhintoq, 2018-12-27:
@smartavionics 
I usually just change the layer height (from .1 to .2) then change it back (.2 to .1)
Different ways to trick it into seeing some change that it will recognize that it needs to re-compute.

smartavionics, 2018-12-27:
This bug (for that is what it is) seems to have come and gone with Cura releases. Some releases have behaved reliably in that changing any setting would make it possible to reslice. This and the last couple of releases have not been so good in this regard.

Ellecross, 2018-12-27:
> 
> 
> @Ellecross
> Yep. I see this _every_ time, _every_ model.
> I have a screen-capture video with real-time clock showing it not changing after 2 minutes 30 seconds. I can't attach an MP4, but if you want to see it just let me know how to send it to you.

No no, I do believe you. Hmm, I would like to have that video and we can try and reproduce it over here. 

@smartavionics I'll raise this issue with the Cura devs when people are back from holidays, I think it needs to have more time for investigation. 

Thank you both for your reports - I saw this before during testing, but it was happening only for me and was worried I was going crazy. :D

Ellecross, 2019-01-03:
Just to update people who see this: 

We're keeping track of this internally. We're trying to reproduce together with system testing and the Cura team. 

If anyone finds an easy way to reproduce this at will, please let us know. 

Thanks guys!

tlhintoq, 2019-01-03:
@Ellecross 
Just an observation, but not a guaranteed "every time" thing...
Yesterday after having had my laptop asleep for several hours with Cura open there were several settings that I could change and not have Cura notice. I had to delete one of the objects on the bed, then control+Z undo the delete, in order to be able to slice.

So maybe a messaging service between the UI and the engine is dieing when the PC is put to sleep for hours while running?

Ellecross, 2019-01-03:
> @Ellecross
> Just an observation, but not a guaranteed "every time" thing...
> Yesterday after having had my laptop asleep for several hours with Cura open there were several settings that I could [...]

Ahh! Good tip! 

I'll try it here and pass the message along to the rest of the testers!

Ellecross, 2019-01-15:
Devs: See CURA-6117 with potential reproduction scenarios

Cura does not slice anymore after per extruder settings have changed. This seems to happen intermittently, but when it happens, it will keep happening.

It does NOT happen on: 

- Global settings
- Platform changes (reload model, resize, disable / enable extruder, etc)