Ultimaker/Cura #5072: [3.6] Missing options on plate adhesion

Liger0, 2018-12-27:
**Application Version**
3.6

**Platform**
Win10 x64

**Printer**
cr-10s double extrusion

**Steps to Reproduce**
Enable the visibility of all the Build plate adhesion settings.

**Actual Results**
Some options are not visible.

**Expected results**
The options should be visible.

**Additional Information**
[image]

Ellecross, 2018-12-28:
Hi @Liger0! 

Good questions! We have gone back to the old tickets we had in Cura to find out the reasons behind this. Thanks to @Ghostkeeper and @Vandrasc 

Those are expected. If you hover over the icon to the left side of the tick box, it says that the setting is hidden by the active machine. 

The Enable Prime Blob setting can be made visible only for the following printers:  (Their printer definition file has the "prime_blob_enable": { "enabled": true } overwrite. ;D)

- builder_premium_large
- builder_premium_medium
- builder_premium_small
- cartesio
- ultimaker3
- ultimaker_s5
- winbo_dragonl4
- winbo_mini2
- winbo_superhelper105

Whilst the Extruder X Y prime position are intentionally hidden since Cura 2.2 for all machines.

Liger0, 2018-12-28:
> Hi @Liger0!
>
> Good questions! We have gone back to the old tickets we had in Cura to find out the reasons behind this. Thanks to @Ghostkeeper and @Vandrasc
>
> Those are expected. If you hover over the [...]

I see, thanks.
So, I will also ask what manages the darker areas on the build plate with double extrusion machine. I noticed using a stock profile the dark area is bigger than using my custom profile, so I guess there is some settings managing it, right?

Ellecross, 2018-12-31:
Yes, there are settings, but Cura also does its own generation of the disallowed areas based on the settings of the profile. 

It's possible to have the "machine_disallowed_areas" setting in the machine definition files. For example, look at Ultimaker 3's (ultimaker3.def.json). The settings are for setting the disallowed areas for the UM3's bed clips, switching bay and for the camera. 

[image]

But Cura also does it's own generation of the disallowed areas depending on the settings of your profile: 

This is with default Prime Blob ON and default Build Plate Adhesion type = Brim. 
[image]

This is with both disabled: 
[image]

closed 2018-12-31 by nallath