Ultimaker/Cura #5075: [3.6] Filament price is not saved

Liger0, 2018-12-28:
**Application Version**
3.6

**Platform**
Win 10 x64

**Printer**
Only tried 1.75 materials

**Steps to Reproduce**
Export a 1.75mm material. Delete it and re import it again

**Actual Results**
The filament cost is set to 0€

**Expected results**
The cost should be saved in the material. 

**Additional Information**
I opened the file and there isn't any referiment to the cost.

Ellecross, 2019-06-27:
Haloo @Liger0, thank you so much for reporting the issues you found. And sorry this has been like, what, 7 months too late. :|

Good thing, this is already fixed in 4.1 and is still working on the current master branch. So, I will close this as it's not an issue anymore. 

P.S. @Vandrasc and I were just talking about how we notice you report issues a lot and that's a good thing. Thank you so much for this! :D :D

closed 2019-06-27 by Ellecross

Liger0, 2019-06-27:
I don't know if the current implementation is to be considered correct.
Now, if I export a material into a file, modify the current material and rename it, then import the file I exported previously, it won't load the details I exported but the details of the material I modified which is now in effect a different material. If I unlink it from that, it resets to 0.

I also have had a weird bug when testing this. 
A material now (the silver in the image) can't be deleted even tho not used by any extruder (the material used by the extruder isn't linked to it too), and 2 materials of same brand name use 2 different dropdowns.
I tried to restart Cura with no luck.
[image]
[cura.zip](https://github.com/Ultimaker/Cura/files/3336785/cura.zip)

Ghostkeeper, 2019-06-28:
Please report these things in separate tickets. One advantage of that is that you'll get a nice template which asks about reproduce steps. I wasn't able to reproduce any of your bugs.

Liger0, 2019-06-29:
Splitted into #5948 and #5949 .