Ultimaker/Cura #5076: Cocoon Create Touch not listed but is Wanhao i3 Plus

iTrooper5118, 2018-12-28:
Hello, I don't know if this is the right way to report a bug or suggestion but under the option for "Other" printers, I noticed there is a Cocoon Create ModelMaker, but no option for Cocoon Create Touch.

The Cocoon Create Touch is more or less a rebranding of Wanhao Duplicator i3 Plus.

Now for those who are familiar with this, they'll select Wanhao Duplicator i3 Plus from the list and print without issues on their Cocoon Create Touch, but for a total newbie to 3D printing, they would never think to select that from the list after not being able to find their printer on the list.

Is it possible to duplicate all the settings from Wanhao Duplicator i3 Plus and create an printer called Cocoon Create Touch from it.

I'm hoping this is an easy one that could be added in 4.0 (or even 3.0?)

Thanks for your patience.

Ellecross, 2018-12-28:
Hey @iTrooper5118, 

We only add new printers - that are not Ultimaker's - if they are submitted via a Pull Request. Like this https://github.com/Ultimaker/Cura/pull/4670 from @pinchies :D 

You can make the pull request yourself, or you can request someone to make it, then we can check it and add it to the next release.

pinchies, 2018-12-28:
@Ellecross - what’s the policy with this? Should both printers be listed separately, and refer to the same profile files, or should the model selection be something like “wanhao duplicator i3 / Cocoon Create Touch”?

iTrooper5118, 2018-12-28:
Hmm, well as long as we don't mix the printers up everything is fine.

So far we have the following printers which are the same but rebadged.

Is there enough space to have it listed as one long description? or is it easier to keep all the Cocoons together in one place, and the Wanhaos in their own group.

Also I know in some other countries, the Wanhaos i3's also go by Balco & Monoprice

Wanhao Duplicator i3 / Cocoon Create 3D Printer
Wanhao Duplicator i3 Plus / Cocoon Create 3D Printer Touch
Wanhao Duplicator i3 Mini / Cocoon Create ModelMaker

iTrooper5118, 2018-12-28:
Here's an image of various Wanhao machines and clones.

[http://itrooper.com/WanhaoClones.jpg](url)

Ellecross, 2018-12-31:
> 
> 
> @Ellecross - what’s the policy with this? Should both printers be listed separately, and refer to the same profile files, or should the model selection be something like “wanhao duplicator i3 / Cocoon Create Touch”?

I don't think it would be a problem if they are separated, but I don't know if they can share the profiles. I can ask the dev team first, but the soonest I'll be able to ask this will be Wednesday (2-Jan). @nallath / @Ghostkeeper :P 

You can also put them together, but we had a problem before with the text display as it didn't fit. 
[image]

and this: 
[image]

That's why it was eventually separated into two in the PR https://github.com/Ultimaker/Cura/pull/4670

pinchies, 2018-12-31:
@Ellecross thanks for the detailed reply. Perhaps a simpler option is to change the setup process to first ask the user for the brand name, and then allow them to select the model name. That would solve several of the problems here, as well as solving the issue of having one very very long list for users to look through.

Ghostkeeper, 2019-01-03:
> Perhaps a simpler option is to change the setup process to first ask the user for the brand name, and then allow them to select the model name. That would solve several of the problems here, as well as solving the issue of having one very very long list for users to look through.

This was the original plan, but Ultimaker's marketing wanted to have a precedence for Ultimaker printers in that menu, for marketing purposes. So for their corporate purposes we've put only Ultimaker in a separate category and the rest under Other.

One thing we could still do is to subdivide Other into subcategories per brand, but that requires some change in the logic (since it's currently all autogenerated from the data in the definition files).

pinchies, 2019-01-03:
That’s exactly what I had in mind. I fully appreciate ultimaker printers taking precedence.