Ultimaker/Cura #5077: Cura 3.6.0 Hangs Freezing on Mac OSX 10.14.2

ngineerken, 2018-12-28:
**Application Version**
3.6.0
**Platform**
Mac OSX 10.14.2

**Printer**
Creality CR10

**Steps to Reproduce**
Update from 3.5.x to 3.6.0

**Actual Results**
Ultimaker Cura Hangs, Freezes if switching to a different app while starting

**Expected results**
Ultimake Cura Just starts FAST

**Additional Information**
It is hanging on Plugins, Profiles etc I have had to Force Quit several times after being nagged to update

ngineerken, 2018-12-28:
I have installed Cura 4.0 Beta, and it loads much faster, but it's still a little rough.
Thanks

ngineerken, 2019-02-24:
Can someone look at improving the 3.6 startup.  I'm seeing others comment that they are having the same issue and even with the 4.x Beta.

Thanks,
Ken

Ghostkeeper, 2019-02-25:
We've made another step towards this for the second beta that we're about to release. I hope that you can test this again. It's a constant effort though.

Maybe you'll find the second 4.0 beta sufficient. First launch of a new version always will be somewhat slower than subsequent launches, since we have to throw away some cached files then.

We won't be updating 3.6 any more unless there is a breaking bug there.