Ultimaker/Cura #5078: [Improvement] Skirt Z distance from first layers

Liger0, 2018-12-28:
**Application Version**
3.6

**Platform**
Win10 x64

**Printer**
Doesn't matter

**Steps to Reproduce**
Use a model which has a different section on the first layers, enable the skirt or brim

**Actual Results**
If the second layer has a bigger sectiion, the skirt/brim will fuse with the model.

**Expected results**
It may be made some checks so Cura sees if the second or n specified layer, like until the third, would intersect with the skirt or brim, to make it bigger and not crossing it in some ways.

**Additional Information**
[image]

Ellecross, 2019-06-27:
This is also fixed now in 4.1 and in current master. I'll close this now @Liger0 
[image]

closed 2019-06-27 by Ellecross

Liger0, 2019-06-29:
I believe you misunderstood.
It wasn't meant to be a bug report but a feature request.
I think it would be nice if the skirt didn't only account for the first model layer but also other n (specified number) of layers, and use it combined with a minimal layer height/height that the model must have z-wise from the skirt, this to avoid the skirt risking to be fused with the model on low layer heights or weird model shapes.

Ghostkeeper, 2019-07-01:
Why did you have "steps to reproduce" on a feature request then :smile: The misunderstanding happened because it was a bug report template, and because we've had a similar issue around November that looked similar to your screenshot.

We're thinking that the cases where the skirt would fail, would fail to print anyway since there would then be at least 3mm overhang on a single layer. Making it put the skirt around multiple layers makes slicing slower and makes the printer use more of the build plate, and makes the skirt less of an indicator for bed levelling than otherwise. So it would make 3D printing worse for most cases while making it better for an edge case that wouldn't print well anyway. That's why we'll reject this feature request.

As a workaround, you can increase the skirt distance.