Ultimaker/Cura #5081: Make elephant foot compensation easier to find

probonopd, 2018-12-29:
**Application Version**
3.6

**Platform**
Linux, probably any

**Printer**
Ender-3, probably any

**Steps to Reproduce**
Search for "Elephant"
[image]

**Actual Results**
Nothing shows up

**Expected results**
An entry field shows up to enter the desired amount of elephant foot compensation

**Additional Information**
https://ultimaker.com/en/blog/50847-introducing-cura-27 talks about "horizontal expansion on initial layer" but it would be good if searching for "elephant" would find this

[image]

probonopd, 2018-12-29:
Note to self: This works for me (0.6 mm nozzle):

[image]

Liger0, 2018-12-30:
It's just that you are using a world invented by yourself.
The correct term is "linear advance", and it's done by enabling it in the firmware and/or Cura marketplace.

probonopd, 2018-12-30:
Elephant foot compensation is about the first layer being to large in the x-y direction. It is not to be confused with Linear Advance, which is about the corners.

Liger0, 2018-12-30:
> 
> 
> Elephant foot compensation is about the first layer being to large in the x-y direction. It is not to be confused with Linear Advance, which is about the corners.

I think you can play with the line width and line height for the first layer to adjust it.

probonopd, 2019-01-11:
> How about adding a feature where the search function also searches the help texts?

I think this is a _great_ idea. I would always type "elephant" because I have already forgotten how the Cura name for this feature is...

Ghostkeeper, 2020-01-17:
There are many ways to deal with elephant foot though, and initial layer horizontal expansion is not usually the best of them. My first approach would be to look at the strength of the walls above the first layers (since quite often the strongest effect is that the skin doesn't shrink but the walls around the infill do). And then I tend to look at the build plate temperature and cooling. And only after that will I look at compensation methods like initial layer horizontal expansion. I think labelling this as "this will fix elephant foot" is misleading.

Ghostkeeper, 2020-01-17:
Maybe it's an idea to include the setting descriptions in the search filter?

probonopd, 2020-01-18:
@Ghostkeeper yes, pretty much so. (I search for "Elephant" in Cura every time...)

mahtDFR, 2020-01-20:
We discussed this morning and think it's a good suggestion. It's on the backlog to be implemented in a future release.

probonopd, 2020-08-25:
Looking at [this video](https://youtu.be/vuKuil0dJqE?t=44), it seems like version 4.7 is my lucky version :-)

Especially like how the video explains the feature with the __elephant foot__ example :+1: 

[image]

Thank you very much Ultimaker Team!

closed 2020-08-25 by probonopd

nallath, 2020-08-26:
It only took us 1.5 year to implement it ;)

Ghostkeeper, 2020-08-28:
It's really effective when combined with the option in the Settings Guide to show the articles in the setting tooltips (which makes it replace the setting descriptions and thus changes what the search function searches through).