Ultimaker/Cura #5082: Slicing don't start

FabLabSNCF, 2018-12-29:
**Application Version**
3.6.0

**Platform**
Windows 7 - 64bits

**Printer**
UM3E

**Problem description**
When my STL file is opened in CURA, slicing never start. I can't export GCODE to file or any drive, I can't use layers view.

[image]

This problem is the same with anothers STL files. Slicing isn't possible.

Liger0, 2018-12-30:
I noticed it happens when, during the start of the slicing, you touch some settings. So you have to  switch the profile and go back to it and slice again.

FabLabSNCF, 2019-01-02:
Hi

It doesn’t fix the problem. I didn’t touch some settings during the start of the slicing…
The problem exists when i’m offline. When internet connexion is OK, slicing can start. It’s verry blocking

Ce message et toutes les pièces jointes sont établis à l'intention exclusive de ses destinataires et sont confidentiels. L'intégrité de ce message n'étant pas assurée sur Internet, la SNCF ne peut être tenue responsable des altérations qui pourraient se produire sur son contenu. Toute publication, utilisation, reproduction, ou diffusion, même partielle, non autorisée préalablement par la SNCF, est strictement interdite. Si vous n'êtes pas le destinataire de ce message, merci d'en avertir immédiatement l'expéditeur et de le détruire.
-------
This message and any attachments are intended solely for the addressees and are confidential. SNCF may not be held responsible for their contents whose accuracy and completeness cannot be guaranteed over the Internet. Unauthorized use, disclosure, distribution, copying, or any part thereof is strictly prohibited. If you are not the intended recipient of this message, please notify the sender immediately and delete it.

Liger0, 2019-01-08:
Okay I am actually having the same problem even when I am online, and it seems permanent.
What happens is really weird. It shows a little advancement on the slicing bar, but then it goes back to 0 and it stays stuck there.
Stopping and trying again doesn't fix it. 
Moving the model and slicing again doesn't work too.
Changing model doesn't work.
Not even restarting the software works.