Ultimaker/Cura #5084: Running from source won't show the UI

gergomez, 2018-12-29:
I'm trying to compile from source (as I already have my own installation of python and everything else and want to do some experiments) but although everything seems to run fine, I only get the Splash screen and the disclaimer, no UI. I've tried with 3.5.1, 3.6.0 and master and all have the same behaviour. I'm running python 3.6.6 and Qt 5.9.2, I've read somewhere that there is a hard dependency on python 3.5.0 but it seems to run fin with 3.6.6, I don't know if that dependency is still relevant (if it is, sorry and just ignore this)

**Application Version**
3.5.1, 3.6.0 and master

**Platform**
Windows x64

**Printer**
n/a

**Steps to Reproduce**
python cura_app.py

**Actual Results**
everything runs no UI

**Expected results**
get a nice UI

**Additional Information**
[cura.log](https://github.com/Ultimaker/Cura/files/2716787/cura.log)

nallath, 2018-12-31:
It should work just fine for python 3.6. I don't see anything in the logs that would explain why it's not showing the interface. 
Do you know what version of pyqt you have installed?

gergomez, 2019-01-11:
In [1]: from PyQt5.QtCore import PYQT_VERSION_STR

In [2]: print(PYQT_VERSION_STR)
5.9.1

gergomez, 2019-01-11:
I've just found that maybe it is related to my version of Qt (5.9.2) I'm going to upgrade to 5.9.7 and check again.

gergomez, 2019-01-11:
Yes, it was QT 5.9.2 (as reported in https://github.com/Ultimaker/Cura/issues/2596)

closed 2019-01-11 by gergomez