Ultimaker/Cura #5085: Pause At Height bug in Cura?

Jaxel, 2018-12-30:
This is a bug I've discovered, and have been able to reproduce... please tell me if anyone else can confirm.

I have a 15 layer STL. Because of the way the layer counter works, that means the layer numbers are actually 0-14, not 1-15. So if I wanted to pause before layer 5, I would actually have to type in 4. Because of this, I instead consider it to be a `Pause AFTER height function`. I am adding the Plugin `Pause at Height`, and I want to pause before layer 11 (with 5 layers remaining). So, going by the 0-14 principle, I would want to set the Pause At Layer to be 10.

However, looking at the gcode, putting in `10` actually pauses at layer 9... But if I put in `11`, it actually pauses at layer 11.

So:
- Pause at 10 = Layer 9
- Pause at 11 = Layer 11

I can't seem to get it to actually pause after layer 10. Can anyone confirm this bug?

uwezi, 2019-03-30:
Version 4.0.0 I have to confirm that there is still something fishy going on with this plugin.
 I will file a new bug report...

Ghostkeeper, 2019-04-01:
Yeah I could also reproduce this, with the following steps:

1. Add Ultimaker 3 printer.
2. Add a cube.
3. Resize the cube to 1.7mm (slicing produces 15 layers). Not sure if this step is necessary.
4. Add the Pause at height post-processing script.
5. Set the script to pause at layer number and fill in the layer 11.

Expected results: M0 command near the `;LAYER:10` comment.
Actual results: M0 command near the `;LAYER:11` comment.

When setting the layer to pause at to 10, it properly pauses near the `;LAYER:9` comment.

Seems like this is an issue of pausing before the layer or after the layer.

uwezi, 2019-04-01:
#5539 was not merely a duplicate - apart from misplacing the pause also the repetition of layers is wrong.

Selecting "0 layers" for repetition gives zero repeated layers
Selecting "1 layer" for repetition results in TWO repeated layers

Ghostkeeper, 2019-04-02:
@nallath said to me via a ticket that he couldn't reproduce this issue. Apparently it doesn't reproduce for all prints. It worked for my 1.7mm cube though (my first try).

uwezi, 2019-04-02:
> 
> 
> @nallath said to me via a ticket that he couldn't reproduce this issue. Apparently it doesn't reproduce for all prints. It worked for my 1.7mm cube though (my first try).

@Ghostkeeper - are you referring to the number of duplicated layers or the off placement of the pause? For me the wrong repetition count is a more severe bug!

ap2649, 2019-11-03:
Hi guys, 

I'm running into this bug too. Cura 4.3. Set up for an Ender 5. Printing at .12mm layer H.

After reviewing the G-code here's what seems to happen:
In Cura Preview the layers start at "1". 
In the G-code the layers start at "0".
In the post process script if I enter layer "6" for the pause, the g-code shows a pause at layer "6".

The problem is that if I am using the Cura Preview to determine what layer I want to pause at then the count is off by 1. 
So, If I want to pause at the start of layer "7" (based on the preview) I need to enter layer "6" in the script. 

This is probably irrelevant in most cases but I ran into this problem when printing a thin part (8 layers in the preview) and want it to pause before the last two layers. When I entered layer "7" in the PPS it was actually pausing at the start of the last layer (last layer in the G-code is "7").

Ghostkeeper, 2019-11-04:
> In Cura Preview the layers start at "1".
In the G-code the layers start at "0".

The post-processing script is already compensating for this, except not really because the post-processing script also gets one additional "layer" as input for the starting g-code so in that code the layer numbers are shifted back, as it were, and it doesn't need to compensate.

The problem that we've observed with this ticket also includes that the layer numbers are shifted by 1 **only** with low layer numbers. At some point (around the 11th layer) it seems to correct itself, which is very strange.

tlhintoq, 2020-02-19:
> Duplicate of #5085.
> The difficulty here is that the off-by-one only exists at certain heights, for some reason.

Disregard height for a bit of testing.  Even if you expressly set the **layer number** - not the height - its still off by one.  So maybe start there.  If you fix the root issue when specifying the exact layer number, then an algorithm built _on top of that_ may be fixed for free. Or may not.  But at the very least if we can get what we expect by layer number its a work around by the user simply looking at the preview layer and going... 
> oh, I want layer 33

tlhintoq, 2020-02-19:
I've always doubted the "at height" could _ever_ be accurate.  If someone enters  2mm, with a layer height of .16mm - There is no exact multiple of .16 that will fall exactly on 2mm - its 12.5 iterations - so the change has to happen at either 12 or 13.    But with layer number, there is no reason for a descripancy even if 'dynamic layer height' is on: You can see see exactly layers in preview.

Ghostkeeper, 2020-04-22:
> I've always doubted the "at height" could ever be accurate. If someone enters 2mm, with a layer height of .16mm - There is no exact multiple of .16 that will fall exactly on 2mm - its 12.5 iterations - so the change has to happen at either 12 or 13.

The intended behaviour is that it should pause just before the first layer that reaches or exceeds the given height. However Z hops mess this sort of thing up so it could be quite complex.

martijnschouten, 2020-09-06:
Alight I spend some time on debugging this issue. Currently a pause command is inserted at the beginning of a layer, if somewhere in that layer the tool is moved above the set pause_height. In that case the tool is moved back to the height that was above pauseheight.

It appears that in the version of cura that I use (see #8305), the tool is moved to the z height of the next layer before next layer itself starts. In #8305 this caused a pause to be inserted one layer to early and the tool to return to a position which is too high. 

I think looking for move to Z commands i not the right way to go to for determining the height at which the printer is printing, because as you said there can be Z hops and so on. A much easier way would be to just calculate the layer number from the z height that the user put in in combination with the layer height and the initial layer height. Then use that to stop the print at the right height.

Also I think what many people will use this feature for is to stop the printer to put in some nuts, or apply some coating or ink in my case. In this situation it is natural to look for the height of the object that you want printed before you pause the print in your cad program and copy that in the pause height. This means that if i put in 1mm, I expect it to pause after the z=1.00 layer has finished printing completely. So at the beginning of the next layer.

I have applied these changes in the following file (currently still with debugging lines):
[PauseAtHeight.zip](https://github.com/Ultimaker/Cura/files/5179677/PauseAtHeight.zip)

martijnschouten, 2020-09-06:
There still was a rounding error in the previous version of the script. Also when relative extrusion was enabled in Cura the pausing script would still return to absolute extrusion mode afterwards. Both have been fixed in this version:
[PauseAtHeight (2).zip](https://github.com/Ultimaker/Cura/files/5180138/PauseAtHeight.2.zip)

Ghostkeeper, 2020-09-07:
> A much easier way would be to just calculate the layer number from the z height that the user put in in combination with the layer height and the initial layer height. Then use that to stop the print at the right height.

That wouldn't work correctly with Adaptive Layers.

martijnschouten, 2020-09-07:
That's true, but in many other cases it is probably a much more robust solution. Maybe somebody knows how to get an array with the adaptive layer heights out of Cura? Then we can do something similar, but with support for Adaptive layers.

Ghostkeeper, 2020-09-10:
It's essentially doing that if you set it to pause at a specific layer number, rather than a height. But even then this bug occurs: https://github.com/Ultimaker/Cura/issues/5085#issuecomment-478570332

closed 2023-06-24 by github-actions[bot]