Ultimaker/Cura #5089: after loading a GCODE file, cannot load STL

WalkerLarry, 2018-12-31:
**Application Version** - 3.6.0

**Platform** - MacOSX Mojave 10.14.2

**Printer** - ZoneStar (nameless)

**Steps to Reproduce**

 - open an GCODE file (the "Layer view" is activated),
 - clear the plate,
 - open an STL file

**Actual Results** - the piece is greyed out and the slicing does not start (it does not start to slice in other views neither)

**Expected results** - not important

**Additional Information**

This is a very minor bug, but it can be used to document more important bugs, linked to some variables that do not change value because of an un-captured exception.

nallath, 2019-01-03:
I've just tried to reproduce this in the 4.0 beta, but i've not been able to do it. Could you perhaps also share your logs?

om2kw, 2019-01-03:
can not reproduce on OSX 10.14.2 with cura 4.0, works fine

smartavionics, 2019-07-06:
Just come across this on master branch Linux. Having loaded a gcode file and then cleared the build plate, on loading an STL file, it no longer gives me the option to slice the model. It looks like the slice button has not been reinstated after the displayed gcode was cleared?

Ghostkeeper, 2019-07-08:
My steps are:

1. (Re)start Cura.
2. Load a g-code file.
3. Press Ctrl+D.
4. Load an STL file.

With those steps still, it doesn't reproduce on this side. There's a big blue slice button waiting to get hit on when I load the STL file.

closed 2020-06-19 by no-response[bot]