Ultimaker/Cura #5091: [3.6.0]RepRap and Makerbot temp commands do not properly set temperatures in dual extrusion mode

mikeshub, 2018-12-31:
**Application Version**
3.6.0

**Platform**
Windows 10

**Printer**
Flash Forge Creator Pro

**Steps to Reproduce**
Setup a two print head job. I used PVA and PLA. After the first tool switch the temp settings start going wrong. The active tool head for temp setting in Sailfish appears to be the last written to and not the active tool head as is the case in Marlin.

**Actual Results**
Among other bad results this ends up having PVA printed at PLA temps.

**Expected results**
The temps to change according to how the slicer is setting them to. 

The below section of gcode highlights the issue:

`T1;

G1 X115 Y70 F7200;

;extruder 1 start code;

M83 ;relative extrusion mode;

M109 S200;

M104 T0 S175;

G1 F1200 E15;

;MESH:hilbert_cube1.stl;

G0 F7200 X93.954 Y65.03 Z0.4;

G0 X93.287 Y62.938;

M104 S210;
`

T0 gets set to 210. The solution is obvious and I would imagine fairly trivial to implement. All temp commands should be accompanied by a tool. This could also be handled by having cura keep track of which tool it has activated for temp commands.

I manually changed the gcodes to be like this and the process worked as the slicer intended:

`T1;

G1 X115 Y70 F7200;

;extruder 1 start code;

M83 ;relative extrusion mode;

M109 T1 S200;

M104 T0 S175;

G1 F1200 E15;

;MESH:hilbert_cube1.stl;

G0 F7200 X93.954 Y65.03 Z0.4;

G0 X93.287 Y62.938;

M104 T1 S210;
`

**Additional Information**
Also I have had to arbitrarily and stuck M104 S{material_print_temperature} in my start gcode because otherwise the slicer inserts M109 S{material_print_temperature} above the start gcode. Seems like there is a little house keeping to be done with multiple nozzle handling. There is no reason for that to be added to my start code that already includes M104 T1 S{material_print_temperature , 1}  and M104 T0 S{material_print_temperature , 0}  That is probably related to the issue discussed in.

#3751 

Thanks a bunch!

nallath, 2019-01-07:
I'm afraid I'm going to have to give the default and somewhat annoying answer; We're probably not going to do this. As is probably already clear from the 500+ open issues, we can't fix all of them. Since this doesn't actually affect Ultimaker printers (which get priority from us when it comes to fixing), it's not likely to be picked up by us.

mikeshub, 2019-01-07:
It's all good. I actually finished porting Marlin 2.0 to the Creator Pro over the weekend. I anticipate finishing the last task of adding RGB led support tonight and plan on making a pull request over there shortly after. I haven't done much in the way of test prints yet, but I don't anticipate any issues with Cura, Marlin, and double nozzles. 

I think we can call this closed. Keep up the good work. It is greatly appreciated.

closed 2019-01-07 by mikeshub

GazparPhen, 2020-10-25:
Were you ever able to get a working profile for this?