Ultimaker/Cura #5097: Cura 4.0 'master' settings

Thisismydigitalself, 2019-01-02:
**Application Version**
4.0 Master

**Platform**
Win10/64
 

I see improvement in regards to settings window which now can be pulled all the way from top to bottom of main workspace but..

- Why the free space on the right to the settings? why wasting real-estate?
- Why Xray view opens in the middle of the screen covering most of the workspace?
- Will we have Expand/Collapse all settings for as quickly as possible jump between settings in different categories? of course best would be what i asked for more than a year now - to see a summery of my prefered settings at all times and being able to click on these for quick tweaking. my work with Cura is mostly tweaking & Re-Slicing many times again and again before printing. 

Still slicing with 3.6 and lovin' it.

[image]
[image]

diegopradogesto, 2019-05-03:
This is fixed in the next 4.1 version, by adding an empty placeholder in between.

[image]

closed 2019-05-03 by diegopradogesto

Ghostkeeper, 2019-05-03:
> Why the free space on the right to the settings? why wasting real-estate?

This is reduced in 4.1 and you can also drag the panel around if you want.