Ultimaker/Cura #5098: A Freecad design is not correctly converted into gcode.

curajdg, 2019-01-02:
**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

curajdg, 2019-01-02:
[image]
[image]

curajdg, 2019-01-02:
Where in Freecad there is a hole of 1 mm in the axis, this is not reproduced in Cura when creating Gcode file.

Please understand, I only know elementary English.

curajdg, 2019-01-03:
> Did you check the model with a model checker? It might be that the model is invalid

OK i try it and give a followup…..

 

JDG

curajdg, 2019-01-03:
Hello,

 

Meanwhile I have found the error.

When exporting to a STL Mesh file I selected too many files.

The error lies with me while working in Freecad.

Thank you for your willingness.

 

Jack DG

curajdg, 2019-01-05:
Hello,

 

Between my initial question and the final answer I can't start Cura anymore.

 

I have uninstalled Cura several times now and installed it again without success.

 

The program works just after the installation but once the program shuts down or the PC turned off the program does not start anymore. 

 

Here I add the dxdiag.

 

Can you help me?

 

Jack DG

 

Van: jack de gendt [mailto:jack.de.gendt@telenet.be] 
Verzonden: donderdag 3 januari 2019 18:15
Aan: 'Ultimaker/Cura'
Onderwerp: RE: [Ultimaker/Cura] A Freecad design is not correctly converted into gcode. (#5098)

 

Hello,

 

Meanwhile I have found the error.

When exporting to a STL Mesh file I selected too many files.

The error lies with me while working in Freecad.

Thank you for your willingness.

 

Jack DG

 

System Information
------------------
      Time of this report: 1/5/2019, 20:56:41
             Machine name: DESKTOP-ME0PKEF
               Machine Id: {4A8653FE-67D8-4D8B-B926-459A81BCACD8}
         Operating System: Windows 10 Pro 64-bit (10.0, Build 17134) (17134.rs4_release.180410-1804)
                 Language: Dutch (Regional Setting: Dutch)
      System Manufacturer: MSI
             System Model: MS-7982
                     BIOS: 3.90 (type: UEFI)
                Processor: Intel(R) Core(TM) i5-6400 CPU @ 2.70GHz (4 CPUs), ~2.7GHz
                   Memory: 8192MB RAM
      Available OS Memory: 8066MB RAM
                Page File: 2556MB used, 6790MB available
              Windows Dir: C:\WINDOWS
          DirectX Version: DirectX 12
      DX Setup Parameters: Not found
[... 834 lines cut ...]
  Microsoft AAC Audio Encoder MFT, {93AF0C51-2275-45D2-A35B-F2BA21CAED00}, 0x1, mfAACEnc.dll, 10.00.17134.0001
  Microsoft ALAC Audio Encoder MFT, {9AB6A28C-748E-4B6A-BFFF-CC443B8E8FB4}, 0x1, MSAlacEncoder.dll, 10.00.17134.0001
  Microsoft Dolby Digital Encoder MFT, {AC3315C9-F481-45D7-826C-0B406C1F64B8}, 0x1, msac3enc.dll, 10.00.17134.0001
Audio Effects:
  AEC, CLSID_CWMAudioAEC, 0x1, mfwmaaec.dll, 10.00.17134.0001
  Resampler MFT, CLSID_CResamplerMediaObject, 0x1, resampledmo.dll, 10.00.17134.0137
Multiplexers:
  Microsoft MPEG2 Multiplexer MFT, {AB300F71-01AB-46D2-AB6C-64906CB03258}, 0x2, mfmpeg2srcsnk.dll, 10.00.17134.0407
Others:
  Microsoft H264 Video Remux (MPEG2TSToMP4) MFT, {05A47EBB-8BF0-4CBF-AD2F-3B71D75866F5}, 0x1, msmpeg2vdec.dll, 10.00.17134.0471

--------------------------------------------
Media Foundation Enabled Hardware Categories
--------------------------------------------

[HKEY_LOCAL_MACHINE\Software\Microsoft\Windows Media Foundation\HardwareMFT]

EnableEncoders = 1
EnableDecoders = 1

-------------------------------------
Media Foundation Byte Stream Handlers
-------------------------------------

[HKEY_LOCAL_MACHINE\Software\Microsoft\Windows Media Foundation\ByteStreamHandlers

nallath, 2019-01-08:
The DixDiag isn't showing much that could help with it. Could you share the Cura logs?

curajdg, 2019-01-10:
The “cura.log” is nowhere to be found on the PC. Not in c and D disk.

 

That may be exactly the problem.

 

The program is installed on my PC on the D disk.

curajdg, 2019-01-10:
Eureka, eureka... suddenly the Cura program starts again by clicking on the same shortcuts.

 

 

Van: jack de gendt [mailto:jack.de.gendt@telenet.be] 
Verzonden: donderdag 10 januari 2019 12:58
Aan: 'Ultimaker/Cura'
Onderwerp: RE: [Ultimaker/Cura] A Freecad design is not correctly converted into gcode. (#5098)

 

The “cura.log” is nowhere to be found on the PC. Not in c and D disk.

 

That may be exactly the problem.

 

The program is installed on my PC on the D disk.

curajdg, 2019-01-10:
Too bad, but I won the victory too early......

Cura doesn't start again.......

 

Van: jack de gendt [mailto:jack.de.gendt@telenet.be] 
Verzonden: donderdag 10 januari 2019 13:07
Aan: 'Ultimaker/Cura'
Onderwerp: RE: [Ultimaker/Cura] A Freecad design is not correctly converted into gcode. (#5098)

 

Eureka, eureka... suddenly the Cura program starts again by clicking on the same shortcuts.

 

 

Van: jack de gendt [mailto:jack.de.gendt@telenet.be] 
Verzonden: donderdag 10 januari 2019 12:58
Aan: 'Ultimaker/Cura'
Onderwerp: RE: [Ultimaker/Cura] A Freecad design is not correctly converted into gcode. (#5098)

 

The “cura.log” is nowhere to be found on the PC. Not in c and D disk.

 

That may be exactly the problem.

 

The program is installed on my PC on the D disk.

curajdg, 2019-01-20:
Cura still does not start after Windows10 is closed.
Every time I have to create a gcode file I have to reinstall Cura.
This problem has only started for a few weeks.
Before that Cura has always worked perfectly.

curajdg, 2019-01-27:
Best,

 

In my Cura maps there is no "cura.log". I find this very strange.

Search with explorer gives the following result.

 

The Cura community comes across to me as very complicated (opaque).

Meanwhile, I have already posed my problem with GitHub (without an answer to date).

That's why I haven't responded to your message before.

I help myself with each time to reinstall Cura.

 

It is not clear to me whether you are part of GitHub.

 

Greetings.

 

Jack De Gendt

Ghostkeeper, 2019-02-14:
You should be able to find the log in: `C:\Users\<your username>\AppData\Roaming\cura\<Cura version>\cura.log` (see also the [readme](https://github.com/Ultimaker/Cura#logging-issues)).

Also keep in mind that attaching files via the e-mail reply functionality in Github will not publish attachments. To post an attachment you will have to go to Github's website.

I'm also curious what you're seeing if Cura doesn't start? Does the splash screen simply disappear? Do you not see a splash screen at all? Does it freeze in the splash screen?

mbartlett21, 2019-06-28:
@curajdg 
@Ghostkeeper 
Would this be the same issue as #5823 ?

Ghostkeeper, 2019-06-28:
Could be but we haven't been able to reproduce either issue so it's impossible to say from here.

luzpaz, 2022-01-24:
anymore traction on this?

fieldOfView, 2022-01-25:
Not without the requested logs. I am a little surprised this issue was not closed due to lack of response to @Ghostkeeper's latest questions here: https://github.com/Ultimaker/Cura/issues/5098#issuecomment-463581418

fieldOfView, 2022-01-25:
The original issue about the freecad design was user error when exporting from freecad (as stated here: https://github.com/Ultimaker/Cura/issues/5098#issuecomment-451212026)

Ghostkeeper, 2022-02-01:
Yeah, reading back at this, this does look as user error due to the UX workflow in FreeCAD, not on Cura's side. I'll close this for that reason.

closed 2022-02-01 by Ghostkeeper