Ultimaker/Cura #5099: 0 Second Print times (Raise N2)

nickarish, 2019-01-02:
**Application Version**
3.6, 4.0
**Platform**
Windows 10/64
**Printer**
Raise3D N2 Dual

**Steps to Reproduce**
Slicing as usual

**Actual Results**
Slicing in Cura produces a 1 hour 49 minute build, but when loaded onto the printer using a usb drive, or when the print file is loaded using ideamaker's open print file setting, it reads a 0 hour, 0 minute, 0 second build.

**Expected results**

Printing

**Additional Information**

[image]

[RN2D_Calibration-Extruder-Offset-6parts-export.zip](https://github.com/Ultimaker/Cura/files/2721234/RN2D_Calibration-Extruder-Offset-6parts-export.zip)

nallath, 2019-01-03:
I'm guessing that the Raise N2 reads the print time from somewhere else than parsing the g-code. 
We don't actually maintain any of the third party profiles (also because we don't actually have the machines to do this) it's not something we can / will fix.

Pull requests to fix this issue are welcome though.

nickarish, 2019-01-03:
When I initially searched the "issues" for Cura I only saw one other issue related to Raise printers. So i'm guessing theres not many people using Cura for them, which is surprising since I haven't been very happy with the results i've gotten from their Ideamaker software. Anyways, i'm new to github, and it seems that most/all people submitting pull requests are writing code and etc, which is something I don't know how to do.