Ultimaker/Cura #5100: Problems with BMP and JPEG import

nyponen, 2019-01-02:
**Application Version**

3.6.0
**Platform**

Windows 10
**Printer**

Ultimaker 2 Go
**Steps to Reproduce**

1. Open a BMP or JPEG file.
2. Click Prepare.
**Actual Results**

The weight of filament needed is given as 0g.
**Expected results**

The correct weight of filament should be given.
**Additional Information**

I print things to spin cast in silver and bronze. In order for the casting to work properly, I need to know the exact weight of the metal needed. If I know the weight of the PLA, I can calculate the weight of metal needed, but Cura doesn't tell me the weight of the PLA.

There are other issues that prevent BMP/JPEG import from being as useful as it could be. For example, the amount of smoothing should be able to be input as a specific number, which would give more reproducible results than using a slider.

nallath, 2019-01-03:
I've just tried to reproduce this with the 4.0 beta, and I get a weight & amount of material used as expected.

nyponen, 2019-01-03:
I just tried this with the 4.0 beta, and got the same problem. The weight is given as 0g.
[image]

nyponen, 2019-01-03:
I get the same problem when I slice an STL. The weight is shown as 0g.

[image]

closed 2019-01-07 by nallath

nyponen, 2019-01-08:
The problem only happens when I specify the Ultimaker 2 Go printer. If I specify Ultimaker 2+, I get the correct filament weight.

fieldOfView, 2019-01-08:
Ah, that makes sense. The Ultimaker 2 Go does not have materials in Cura. Materials are selected on the printer, so Cura does not know what material is used. Because Cura does not know what material is used, it cannot calculate the weight.

nyponen, 2019-01-08:
The Ultimaker 2 Go only supports PLA. You cannot select any other material on the printer.

fieldOfView, 2019-01-08:
What you could do is go to "Machine Settings" (on the Printers pane of the preferences) and select "Marlin" instead of "Ultimaker 2" as the GCode flavor for your printer.