Ultimaker/Cura #5103: Printing a model with the 2nd extruder only causes unnecessary switch commands

zeleps, 2019-01-03:
**Cura version 3.6.0**

**Steps to Reproduce**
Save g-code for a model using only extruder 2 (for model, skirt, support, no prime tower) . Set **1st nozzle** switch retraction distance (in my case at 16mm), and some **extruder 1** end G-Code.

**Actual Results**
Before skirt printing begins, although extruder 2 is selected from the very beginning (T1 command), commands for switching to extruder 2 are added to the file (erroneously), resulting in non-revocable retraction of extruder 2, plus any extruder 1 end commands. This is easily detected in the g-code file.

**Expected results**
No switching commands should be executed.

**Additional Information**
The following g-code clearly shows the problem (my comments begin with ;;;):
```

;FLAVOR:Marlin
;TIME:1174
;Filament used: 0m, 1.34502m
;Layer height: 0.2
;Generated with Cura_SteamEngine 3.6.0
T1 ;;; This is correctly included, so active extruder is now E1
M140 S57
M105
M190 S57
M104 T1 S215
M105
M109 T1 S215
...
;;; my printer's start g-code, irrelevant
...
G92 E0
G1 F1500 E-3
;LAYER_COUNT:101
;LAYER:0
;;;The following 4 lines shouldn't have been included in the file:
G1 F1200 E-16 ;;; retraction set for E0 nozzle switch, but E0 was never active, so this is executed on E1 and that's a problem
G92 E0
G1 F3800 X190 Y170 ;;; my E0 exit g-code
T1 ;;; Does nothing, E1 was already active, but resets E1 position, so -16 is now 0.
G92 E0
M107
G0 F3600 X86.577 Y87.328 Z0.1
;TYPE:SKIRT
... ;;; rest of the code is fine

```

As a result of the above issue, the skirt and part of the first layer are not printed, since the filament was never primed correctly.

rburema, 2019-01-09:
Devs: CURA-6093

Turns out the `gcodeExport` `current_extruder` defaults (in `preSetup()`)  to `0` instead of `start_extruder`, which all callers know.

rburema, 2019-01-10:
Resolved by (pending Q&A): https://github.com/Ultimaker/CuraEngine/pull/974

closed 2019-01-10 by rburema