Ultimaker/Cura #5106: Cura with firmware retraction on does not generate M207/M208 commands

ethanspitz, 2019-01-04:
**Application Version**
3.6.0 with Printer Settings Plugin

**Platform**
Windows 10

**Printer**
CR-10S Pro
[CCR10S_Retraction.curaproject.3mf.zip](https://github.com/Ultimaker/Cura/files/2725962/CCR10S_Retraction.curaproject.3mf.zip)

**Steps to Reproduce**
1. Enable FW Retraction using the Printer Settings Plugin.
2. Inspect generated GCODE.

**Actual Results**
Cura's retraction settings are not sent to the printer using the M207/M208 commands

**Expected results**
Cura should add M207/M208 commands to the generated GCODE.

**Additional Information**
I actually think there are a couple ways to skin this.

1. Add an "Enable Retraction Control" option similar to "Enable Acceleration Control" when FW Retraction is enabled. With this option enabled, M207/M208 would be in the GCODE. Without it, only G10/G11 with no overrides.
2. Hide Retraction related settings that don't get used when FW Retraction is enabled
3. Always send M207 and M208
4. Do nothing.

Obviously, I don't think 4. is the right thing to do. In my opinion, 1. would be the best thing to do as it mirrors the other firmware overrides of acceleration and retraction and is least confusing to the user.

nallath, 2019-01-07:
Option 4 is probably the option that's going to happen since Ultimaker printers don't use the m207 or m208 (they just use the G10/G11). As we unfortunately already have way too many issues (and those that affect/improve Ultimakers do get priority from us), it's not that likely that we will implement this. 

You could add the M207 / M208 to your start g-code as a quick fix for this.

darconeous, 2020-07-10:
I can *almost* get this working with the following start gcode:

```
M207 F{retraction_speed} S{retraction_amount} Z{retraction_hop}
M208 F{retraction_prime_speed} S{retraction_extra_prime_amount}
```

However, the problem is that the units for the feedrate (`F`) field are mm/min, whereas `retraction_speed` and `retraction_prime_speed` are in mm/sec. Argh, so close!

I tried doing something fancy like `F{retraction_speed * 60}`, but predictably that did not work. A possible work-around would be to add something like this to `fdmprinter.def.json`...

```
"retraction_feedrate": {
    "label": "Retraction Feedrate",
    "description": "The retraction speed in mm/min",
    "unit": "mm/min",
    "type": "float",
    "value": "retraction_speed * 60",
    "default_value": 1500,
    "minimum_value": "0.0001",
    "settable_per_mesh": false,
    "settable_per_extruder": false,
    "settable_per_mesh_group": false
},
```

tommylux, 2021-02-07:
I'd also like to set the firmware retraction in start code. Does start code support mathematical functions? I assume not based on darconeous's comments.

fvrmr, 2021-02-12:
Hi @tommylux no the start gcode doesn't support mathematical functions. And like @darconeous said you could add it in your `fdmprinter.def.json`.

tommylux, 2021-02-20:
Some more explanation required and background regarding the work-around for modifying the fdmprinter.def.json.

1. How is this file associated with printers.: Would modifying this file affect all printers or only new printers created going forward? This directory appears to have all sorts of settings for different printers. If you create a new printer, which printer to do you select?
3. retraction_feedrate does not exist, which branch should it be added to.
4.  Does this replace the M207 F{retraction_speed} S{retraction_amount} Z{retraction_hop}. or do you need to now reference the new variable ie, : M207 F{retraction_speed} S{_retraction_feedrate_} Z{retraction_hop}

Thanks.

tommylux, 2021-02-21:
I tell you what might be a suitable work-around. Simply enter a retraction amount multiplied by 60 in the Retraction Distance field. When M207 is sent, your modified value * 60 will be inserted in the M207 command.

Ghostkeeper, 2021-03-01:
For clarity, the proposed workaround was to edit `.def.json` files, not necessarily `fdmprinter`.

In the printer definition files (extension `.def.json`), the `"value"` property of a setting can be an arbitrary Python expression, allowing you to make calculations. For the start g-code, the result needs to be a string. So to construct a string of start g-code with calculations inside, you can do something like this:

```
"machine_start_gcode": {
    "value": "\"G92 E0\nM207 F\" + str(retraction_speed * 60) + \" S{retraction_amount} Z{retraction_hop}\""
},
```

> How is this file associated with printers.: Would modifying this file affect all printers or only new printers created going forward? This directory appears to have all sorts of settings for different printers. If you create a new printer, which printer to do you select?

A `.def.json` file is associated with a single printer model. The total list of printer definitions is seen when you add a new printer to Cura and you have to select from the offline printers. Only, some of the printers are invisible in that list, and some printers inherit from other printers. As it happens, all printers ultimately inherit from `fdmprinter`, just modifying some overrides to specify their own width, height, and other settings. So effectively, if you edit `fdmprinter`, you'll edit the default values for all printers, except that some printers may override your defaults if they think they're special in some aspect. The start g-code is particularly popular to edit so I'd expect that modifying that in `fdmprinter` will not have any effect on most printers.

> retraction_feedrate does not exist, which branch should it be added to.

If this is intended to be a new feature in Cura and you want to make a pull request to add it, point the pull request at Master. We add new features into Master.

> Does this replace the M207 F{retraction_speed} S{retraction_amount} Z{retraction_hop}. or do you need to now reference the new variable ie, : M207 F{retraction_speed} S{retraction_feedrate} Z{retraction_hop}

The proposed workaround is to write the start g-code in a different spot where you can make calculations (the inner configuration files of Cura itself). So you could say it replaces it, yes. But really you'll just have a start g-code that automatically adjusts itself depending on other settings.

> I tell you what might be a suitable work-around. Simply enter a value multiplied by 60 in the Retraction speed field, ignoring the mm/s. When M207 is sent, your modified value * 60 will be inserted in the M207 command.

That's a good workaround, too! After all, when firmware retracts are enabled, it shouldn't really use the retraction speed any more, should it? I'd say that exploits a "bug" in Cura (which is maybe not so serious) because normally settings that are not used will not be displayed to the user either.

tommylux, 2021-03-02:
These files seem to allow you to define which settings are applied to all extruders. Since I usually intend my configuration changes to apply to all extruders, I wonder if I can make the configuration items all linked by editing these files. (Currently I export the profile and extract and copy the configuration items to the global file and re-import)

Another query is whether I can define new settings to appear which I use in my own reference.. for example just exposing a new field for retraction-feedrate, I can then reference that field in the start code and then just type the feedrate amount ie 2700 for 45m/s. Though perhaps this won't work if it's not in the list of exposed keywords.

One thing I realised was setting a retraction speed as the feedrate may affect other settings, such as octoprint timelapse which requires this input.

war4peace, 2021-05-25:
If you have RRF firmware, in the G-Code dictionary, M208 is for Set axis max travel.
https://duet3d.dozuki.com/Wiki/Gcode#Section_M208_Set_axis_max_travel

Make sure you don't use it like in the example above if you have RRF.

JohnTasto, 2021-06-26:
I'm not able to get this to work. If I do a direct copy and paste of Ghostkeeper's code sample into my machine definition, the printer settings dialog just says `None` in the G-code textarea, and when I try to slice, I get "The following settings have errors: Start G-code", and the following written to the logs:
```
2021-06-25 20:04:03,132 - WARNING - [MainThread] UM.Logger.logException [113]: Exception: Could not validate setting machine_start_gcode, an exception was raised
2021-06-25 20:04:03,135 - WARNING - [MainThread] UM.Logger.logException [117]: Traceback (most recent call last):
2021-06-25 20:04:03,139 - WARNING - [MainThread] UM.Logger.logException [117]:   File "C:\Users\burto\cura\cura-build\build\inst\lib\python3.8\site-packages\UM\Settings\Validator.py", line 75, in __call__
2021-06-25 20:04:03,143 - WARNING - [MainThread] UM.Logger.logException [117]: ValueError: Cannot validate None, NaN or similar values in setting machine_start_gcode, actual value: None
```
Note, I am running @smartavionics' modified version of Cura and a bunch of plugins. I've attached my configs if anyone is interested. They are a bit more complicated than when I got the above errors, when I had them pared down to the bare minimum trying to figure it out. In particular I'm doing the hardware retract stuff in `machine_extruder_start_code` instead of `machine_start_gcode`, but the errors are practically the same.

Related question: Is it possible to use this technique in `machine_extruder_start_code` and somehow reference the current layer to make it even more dynamic? Specifically I want to get `retraction_hop_initial_layers` working with hardware retraction.

[bibent.zip](https://github.com/Ultimaker/Cura/files/6719547/bibent.zip)

Ghostkeeper, 2021-07-05:
> I'm not able to get this to work. If I do a direct copy and paste of Ghostkeeper's code sample into my machine definition, the printer settings dialog just says `None` in the G-code textarea, and [...]

Sorry, my example was meant to be illustrative. I didn't test it. Turns out I forgot a backslash before the `\n`, or JSON would unescape the `\n` directly which would be a syntax error. Try it again with my edit. I tested that and it works fine, producing this line in the BIBO Ditto of your definition files:

    M207 F1500 S{retraction_amount} Z{retraction_hop}

The retraction speed there is the global retraction speed of 25, which is not the one you see in the interface nor the one being used by the print while extruder 1 is active. You'd have to use `extruderValue(0, "retraction_speed")` instead to get it from the first extruder, yada yada, but the idea still works.

> Related question: Is it possible to use this technique in `machine_extruder_start_code` and somehow reference the current layer to make it even more dynamic?

You can't make the extruder start g-code change per layer, if that's what you're asking. It's evaluated once, in the Python environment of Cura's front-end, before layers even exist. CuraEngine can't execute those Python formulas so trying to make it change per layer with a formula is quite far out of reach.