Ultimaker/Cura #5109: cura 3.6.0 network drives are not shown under Fedora 29

Stoccarda, 2019-01-04:
**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

nallath, 2019-01-07:
I fear this is an upstream bug from Qt. We might just have to wait until it's fixed there.

Stoccarda, 2019-01-07:
Sorry for may have confusing with the caption:
Network drives are neither visible in 3.5.1 nor 3.6.0 nor 4.0.0, only in the Fedora 'files' program.
But besides of this, the Open Dialogue has changed after 3.5.1, local drives and removable media (e.g. the USB stick) are not directly visible.
If it is Qt related (newer version after 3.5.1) , then ...

nallath, 2019-01-08:
Well, I'm on fedora 28 and I do see the local drives / removable media. We used to have our own implementation of the file dialog, but switched back to the system provided one since we had quite a few issues with it.

Stoccarda, 2019-01-08:
Do you also the difference between 3.5.1 and 3.6.0 / 4.0.0 beta?

nallath, 2019-01-08:
Nope, but as I'm running it from source, I'm always using the same QT version.

Stoccarda, 2019-01-08:
And using the download versions?