Ultimaker/Cura #5111: 3.6.0 Wringly turning off nozzle heating in a dual color, mixing extruder print

Le-zOU, 2019-01-04:
**Application Version**
3.6
**Platform**
not relevant

**Printer**
Geeetech A20M

**Steps to Reproduce**
Print a dual color model.
as soon as 1 color is done and only the other is left to print, nozzle temp turns off, resulting in the 2nd colour being unable to be extruded

**Actual Results**
2nd color filament doesn't extrude

**Expected results**
2nd color filament should extrude, single nozzle temp shouldn't be set to 0 until the entire print is finished

**Additional Information**

Gcode
[A20M_Starfish.zip](https://github.com/Ultimaker/Cura/files/2727573/A20M_Starfish.zip)

Le-zOU, 2019-01-04:
maybe duplicate of: #https://github.com/Ultimaker/Cura/issues/2920

nallath, 2019-01-07:
The same is unfortunately still true; Cura doesn't have actual support for mixing extruder. 

What you can do is set the material_standby_temperature to be the same as your printing temperature.

closed 2019-01-07 by nallath

Le-zOU, 2019-02-16:
Why was that closed ? Any chance of a fix on 3.6.* or 4.* ?
Thank you

Ghostkeeper, 2019-02-18:
It was closed as a duplicate. There is a workaround for it: Set all temperatures to be equal. That includes the stand-by temperature, the initial printing temperature, the final printing temperature, etc.

Let's keep this issue open as the "original" of all the duplicates: https://github.com/Ultimaker/Cura/issues/4204

Le-zOU, 2019-02-18:
That workaround doesn't work as all my temp are already set at the same value

Ghostkeeper, 2019-02-18:
So what temperature does it set your nozzle to then, if all temperatures are at the same value?
Here's a project file and the resulting g-code. No M104/M109 commands occur except at the beginning and the end of the print. And once it sets it unnecessarily to 200 degrees again, but of course if your temperatures are all the same, it's only going to be able to set it to the same temperature.

[test.zip](https://github.com/Ultimaker/Cura/files/2874993/test.zip)

Are you sure that your temperatures are all the same even on the other extruders that you're using in Cura?

Le-zOU, 2019-02-18:
I've printed a Phil  dual color and a flexible starfish (gcode provided in the 1st post)
https://www.thingiverse.com/thing:2849214
https://www.thingiverse.com/thing:3273805

After the 2nd color is done, extruder target temp drops to 0.
So the last layers of color1 can't finish as filament doesn't extrude.

All my temps are set to 200, on all extruders.
[image]

Phil Gcode
[Phil_Color.zip](https://github.com/Ultimaker/Cura/files/2875091/Phil_Color.zip)

line 852681 M104 T1 S0

Le-zOU, 2019-02-18:
printing your test gcode (added G28 as it was missing)

Ghostkeeper, 2019-02-18:
Ah, I see what you mean.

Then maybe you'd have to use a search-and-replace post-processing script that matches on M104 and M109 and removes those? And then add a heating command to the start g-code that the regex doesn't match on.

Le-zOU, 2019-02-18:
Yes, but the point is that shouldn't happen in the first place :)

Ghostkeeper, 2019-02-18:
Indeed. But of course for other printers it should cool down the nozzles that it's no longer using.

We don't have those printers here at work to test with, we don't have an emulator and we don't have the time to support all printers out there. We would accept a PR as long as it doesn't break the workflows of other printers, but we can't really do anything more than that.

Le-zOU, 2019-02-18:
I understand.
It is a mixing hotend so 2 extruder motors 1 hotend.
Maybe there's a way of declaring that "mixing setup" somewhere so Cura knows there's only 1 hotend and behaves accordingly ?

Ghostkeeper, 2019-02-18:
Not currently, so it'd need to be implemented.

MB3Y, 2019-02-19:
Linking #2918 too.